PX4/PX4-Autopilot #15291: make px4_sitl_default gazebo fail (fatal error: gst/gst.h: No such file or directory)

liganggg, 2020-07-08:
hi
i am using ubuntu 16.04 with gazebo 9
px4 version is 1.10.2
when i px4_sitl_default gazebo 
it report fail 

`FAILED: /usr/bin/c++   -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -Dgazebo_gst_camera_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /home/l/anaconda3/include/qt -isystem /home/l/anaconda3/include/qt/QtCore -isystem /home/l/anaconda3/./mkspecs/linux-g++ -I/home/l/px4/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/l/px4/Firmware/mavlink/include -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv -I/home/l/px4/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++14 -MMD -MT CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -MF CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o.d -o CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -c /home/l/px4/Firmware/Tools/sitl_gazebo/src/gazebo_gst_camera_plugin.cpp
In file included from /home/l/px4/Firmware/Tools/sitl_gazebo/src/gazebo_gst_camera_plugin.cpp:24:0:
/home/l/px4/Firmware/Tools/sitl_gazebo/include/gazebo_gst_camera_plugin.h:31:21: fatal error: gst/gst.h: No such file or directory
compilation terminated.`

it seem like loss some file
please give me some suggestion, tks!

Jaeyoung-Lim, 2020-07-08:
You need to install gstreamer. please follow the startup guide in dev.px4.io

liganggg, 2020-07-08:
> You need to install gstreamer. please follow the startup guide in dev.px4.io

thanks for you advice,

I had solved this problem by use 
sudo apt-get install libgstreamer1.0-dev

closed 2020-07-08 by liganggg