PX4/PX4-Autopilot #15294: make px4_sitl_default gazebo Error

Robotmaid, 2020-07-09:
I am using ubuntu 18.04 with gazebo 9 and ros-melodic
When I run 
```
git submodule update --init --recursive
git checkout v1.11.0-beta1
make distclean
make px4_sitl_default gazebo
```
It report Error :
```
make: Warning: File ‘Makefile’ has modification time 22533 s in the future
— PX4 version: v1.11.0-beta1
— PX4 config file: /headless/Desktop/catkin_ws/Firmware/boards/px4/sitl/default.cmake
— PX4 config: px4_sitl_default
— PX4 platform: posix
— PX4 lockstep: enabled
— cmake build type: RelWithDebInfo
— The CXX compiler identification is GNU 7.5.0
— The C compiler identification is GNU 7.5.0
— The ASM compiler identification is GNU
— Found assembler: /usr/bin/cc
— Check for working CXX compiler: /usr/bin/c++
— Check for working CXX compiler: /usr/bin/c++ — works
— Detecting CXX compiler ABI info
— Detecting CXX compiler ABI info – done
— Detecting CXX compile features
— Detecting CXX compile features – done
— Check for working C compiler: /usr/bin/cc
— Check for working C compiler: /usr/bin/cc — works
— Detecting C compiler ABI info
— Detecting C compiler ABI info – done
— Detecting C compile features
— Detecting C compile features – done
— Building for code coverage
— ccache enabled (export CCACHE_DISABLE=1 to disable)
— Found PythonInterp: /usr/bin/python3 (found suitable version “3.6.9”, minimum required is “3”)
CMake Error at src/lib/CMakeLists.txt:48 (add_subdirectory):
The source directory

/headless/Desktop/catkin_ws/Firmware/src/lib/ecl

does not contain a CMakeLists.txt file.

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_airdata”.
Call Stack (most recent call first):
src/modules/airspeed_selector/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo_lookup”.
Call Stack (most recent call first):
src/modules/attitude_estimator_q/CMakeLists.txt:34 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/commander/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_EKF”.
Call Stack (most recent call first):
src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/ekf2/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_l1”.
Call Stack (most recent call first):
src/modules/fw_pos_control_l1/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_tecs”.
Call Stack (most recent call first):
src/modules/fw_pos_control_l1/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/local_position_estimator/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/mavlink/CMakeLists.txt:36 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/mc_pos_control/CMakeLists.txt:37 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/navigator/CMakeLists.txt:34 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_l1”.
Call Stack (most recent call first):
src/modules/rover_pos_control/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_validation”.
Call Stack (most recent call first):
src/modules/sensors/CMakeLists.txt:41 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/simulator/CMakeLists.txt:56 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo”.
Call Stack (most recent call first):
src/modules/vmount/CMakeLists.txt:33 (px4_add_module)

CMake Error at cmake/px4_add_module.cmake:205 (get_target_property):
get_target_property() called with non-existent target “ecl_geo_lookup”.
Call Stack (most recent call first):
src/systemcmds/tests/CMakeLists.txt:88 (px4_add_module)

CMake Error at /usr/share/cmake-3.10/Modules/ExternalProject.cmake:2474 (message):
No download info given for ‘sitl_gazebo’ and its source directory:

/headless/Desktop/catkin_ws/Firmware/Tools/sitl_gazebo

is not an existing non-empty directory. Please specify one of:

* SOURCE_DIR with an existing non-empty directory
* DOWNLOAD_COMMAND
* URL
* GIT_REPOSITORY
* SVN_REPOSITORY
* HG_REPOSITORY
* CVS_REPOSITORY and CVS_MODULE
Call Stack (most recent call first):
/usr/share/cmake-3.10/Modules/ExternalProject.cmake:3029 (_ep_add_download_command)
platforms/posix/cmake/sitl_target.cmake:33 (ExternalProject_Add)
platforms/posix/CMakeLists.txt:115 (include)

— Configuring incomplete, errors occurred!
See also “/headless/Desktop/catkin_ws/Firmware/build/px4_sitl_default/CMakeFiles/CMakeOutput.log”.
Error: /headless/Desktop/catkin_ws/Firmware/build/px4_sitl_default is not a directory
Makefile:198: recipe for target ‘px4_sitl_default’ failed
make: *** [px4_sitl_default] Error 1
```

julianoes, 2020-07-10:
After you do `git checkout v1.11.0-beta1` you need to do `git submodule update --init --recursive` again.

Robotmaid, 2020-07-10:
> After you do `git checkout v1.11.0-beta1` you need to do `git submodule update --init --recursive` again.

It's still reporting the same Error.

julianoes, 2020-07-14:
It complains about a `CMakeLists.txt` not being there. Can you check if it is actually there.

And can you paste the output of `git status`?

closed 2021-02-02 by Jaeyoung-Lim