PX4/PX4-Autopilot #15295: make px4_sitl gazebo fail

yaseminarpaci, 2020-07-09:
after sudo make px4_sitl gazebo command this fail comes,

FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/yasemin/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/yasemin/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/yasemin/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_sitl] Error 1

Jaeyoung-Lim, 2020-07-09:
@yaseminarpaci Please post the full log

yaseminarpaci, 2020-07-09:
sudo make px4_sitl gazebo
[sudo] password for yasemin: 
[0/6] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
CMake Error at cmake/EnableC++XX.cmake:26 (message):
  The compiler /usr/bin/c++ has no C++11 or above support.  Please use a
  different C++ compiler.
Call Stack (most recent call first):
  CMakeLists.txt:44 (include)

-- Configuring incomplete, errors occurred!
See also "/home/yasemin/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/yasemin/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[1/6] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/yasemin/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -GNinja /home/yasemin/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/yasemin/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:226: px4_sitl] Error 1

Jaeyoung-Lim, 2020-07-09:
@yaseminarpaci You shouldn't be using sudo. Please follow the documentation written in https://dev.px4.io/master/en/simulation/gazebo.html

Jaeyoung-Lim, 2020-10-12:
@yaseminarpaci Please reopen if this issue persists. Closing since stale

closed 2020-10-12 by Jaeyoung-Lim