PX4/PX4-Autopilot #15305: STM32H7 CAN driver UAVCAN GlobalTimeSyncMaster pub conflict

mirkix, 2020-07-10:
**Describe the bug**
When using a STM32H7 based autopilot like the HEX CubeOrange with UAV-CAN it publish `ERROR UAVCAN GlobalTimeSyncMaster pub conflict` messages.

**To Reproduce**
Steps to reproduce the behavior:
1. Load PX4 master on a CubeOrange
2. Enable UAV-CAN `param set UAVCAN_ENABLE 2`
3. Connect SLCAN Interface to CubeOrange
4. Start UAVCAN GUI Tool
5. Get Error Message in UAVCAN GUI Tool

**Log Files and Screenshots**
[image]

mirkix, 2020-07-10:
@JacobCrabill

JacobCrabill, 2020-07-15:
Hmm, I can give this a look next week perhaps.

JacobCrabill, 2020-07-20:
Just took a brief look at it; this comment in GlobalTimeSyncMaster may hold the key to the issue:

```
/**
 * Please read the specs to learn how the time synchronization works.
 *
 * No more than one object of this class is allowed per node; otherwise a disaster is bound to happen.
 *
 * NOTE: In order for this class to work, the platform driver must implement
 *       CAN bus TX loopback with both UTC and monotonic timestamping.
 *
 * Ref. M. Gergeleit, H. Streich - "Implementing a Distributed High-Resolution Real-Time Clock using the CAN-Bus"
 *
 * TODO: Enforce max one master per node
 */
```

Specifically: **NOTE: In order for this class to work, the platform driver must implement CAN bus TX loopback with both UTC and monotonic timestamping.**

I don't know when I'll have free time to look into this, but this could give another developer a clue on where to start digging into the driver.

PR #15478 (2020-08-05): uavcan: stm32h7: Fix improper loopback handling
Fixes #15305 

FYI @mirkix
commit b1641740 (2020-08-05): uavcan: stm32h7: Fix improper loopback handling

Fixes #15305
closed 2020-08-05 by dagar