PX4/PX4-Autopilot #15312: Pixhawk 4 + Drotek Sirius RTK GNSS compasses inconsistent

nrogelio, 2020-07-12:
**Drone**
S500 with Pixhawk 4 and Drotek Sirius RTK GNSS (with RM3100 magnetometer).
Pixhawk 4 with rotation_roll 180
GPS no rotation

After loading commit e819f99 firmware to Pixhawk 4, the vehicle can't arm due to compasses inconsistent pre-arm check error, this error is not cleared even after compass calibration. Pixhawk 4 and GPS (with external compass) are facing forward, but the error says they are inconsistent.

Compass calibration can be performed without issue, but this will not clear the pre-arm error.

Compass calibration was performed outdoors and indoors with the same result.

Checked with the GPS (with compass) included with Pixhawk 4 and is working, so this is issue is with the Drotek Sirius RTK GNSS device. 

**To Reproduce**
1. Use Pixhawk 4 + Drotek Sirius RTK GNNS
2. Load commit  e819f99 or master
3. Try to arm
4. See compasses inconsistent error
5. Calibrate compass
6. Try to arm
7. See error

**Expected behavior**
Vehicle should be able to arm since compasses are facing same orientation.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=8e97f377-2aff-4c9a-8306-a9f13b43df1c

[image]

dagar, 2020-07-29:
Could you turn on logging from boot plus "sensor comparison" and go through the mag calibration? If possible also connect the default Holybro GPS/mag puck (properly orientated) as another comparison.

Then I'll compare all the mag data side by side per axis and see if I can figure out what's going on. Posting pictures of the setup and any details about the Drotek GPS with RM3100 might also help.

lukegluke, 2020-07-29:
Hi @dagar,
Sorry for intervention, with a not really related issue, but I have similar compasses inconsistent on our custom rover board with ist8310 and rm3100 compasses. I've checked them separately: ekf with ist8310 works very well, but with the rm3100 board is heading in the wrong direction (about -100°) and also after sharp movements the compass in qgc starts slowly floating.
I've logged the calibration process of our board as you suggested to @nrogelio. As I see y data of two mags are matched (mag0 is rm3100), while x and z axes are opposite.
[image]

Is this sensor orientation issue? rm3100 doesn't have any rotation on my board, but I still wonder is rotation from driver argument -R is performed before or after any axes swaps in driver internally like in collect() of ist8310 for example?

Could you please advise a little where to look, how to properly interpret these data?
Big thanks in advance!

dagar, 2020-07-29:
Thanks for the additional info @lukegluke. I should probably get one of these units and take a closer look at the driver.

nrogelio, 2020-07-29:
Hi @dagar ,

I configured logging from boot to shutdown and "sensor comparison".

Loaded commit f7415c3 from master branch and No compasses inconsistent issue was found, neither with 2 external compasses connected (RM3100 + Holybro GPS/mag) or with only 1 external compass connected (RM3100)

master_f7415c3_2_external_mags_Passed
https://review.px4.io/plot_app?log=dbdb9fa5-7f23-4b40-bb00-135600e43793

master_f7415c3_1_external_mag_rm3100_Passed
https://review.px4.io/plot_app?log=7b6444c0-40b1-45cc-93cd-d278bd4f6e8d

Loaded commit e819f99 and used only RM3100 external compass, I was able to reproduce the compasses inconsistent issue.

master_e819f99_1_external_mag_rm3100_Failed
https://review.px4.io/plot_app?log=2b0a5329-ccd9-4d94-a329-dd298d22fc58

Connected the 2 external compasses (RM3100 and Holybro mag) and NO compasses inconsistent issue was found.

master_e819f99_2_external_mags_Passed
https://review.px4.io/plot_app?log=44a8335b-1910-48d8-86c3-2768647083d3

To summarize:
e819f99 - RM3100 only - Failed
e819f99 - RM3100 + Holybro mag - Passed
f7415c3 - RM3100 only - Passed
f7415c3 - RM3100 + Holybro mag - Passed

Here is my setup:
[image]
Pixhaw 4 is on the bottom with rotation roll 180

Here you can find pictures of the Drotek GPS:
Top
[image]

Bottom
[image]

Pins tracking
[image]

lukegluke, 2020-07-30:
> In those plots does your board (and internal ist8310 mag) have any rotation? The configurable external rotation parameter is applied after `sensor_mag` in the `sensors` hub.

Board doesn't have rotation. ist8310 mag driver starts with -R 12 (Pitch 180°) rotation. I also had imu icm-20689 - driver starts with -R 2 parameter (Yaw 90°).

While rechecking sensor orientations I notice that it looks like we have bug on our board - input and output pads of Z and X coils are switched.
[image]

As I understand it leads to exactly this behavior - x and z data switch sign, right?
I add virtual rotation -R 12 (Pitch 180°) to flip x and z axes and after calibration got right result without inconsistency between compasses. So my case seems to be only hardware problem of custom board, sorry for bothering!

lukegluke, 2020-07-30:
By the way, in the @nrogelio picture of Drotek GPS rm3100 Z coil is placed not as in manual picture - along x with polarity indicator and ZINN facing forward - is it right to do so?

nrogelio, 2020-08-02:
The spec mentions: "The Sen-XY-f is insensitive to the location of the polarity indicator, while the location of the polarity indicator is critical for the Sen-Z-f." so X and Y can be connected inverted and shouldn't cause a problem, and to me Sen-Z-f is also polarized properly, but it's confusing.

lukegluke, 2020-08-02:
> "The Sen-XY-f is insensitive to the location of the polarity indicator, while the location of the polarity indicator is critical for the Sen-Z-f." so X and Y can be connected inverted and shouldn't cause a problem

I suppose it's different things - switch pads and flip coil. Sensor has to distinct direction of magnetic field, so I really think that pads position gives positive direction and it's important.
_"The coil’s inductance is a function of the magnetic field strength. As such, the frequency of oscillation of the circuit varies with the strength of the total magnetic field parallel to the sensor. To make a measurement, one side of the coil is
grounded while the other side is alternately driven with positive and negative current through the oscillator."_

Paul-DROTEK, 2020-08-05:
Hi there,
This is Paul from Drotek's team here.

X and Y coils are indeed insensitive to the polarity because both coils are winded the same way (basically the current that goes from the P pole to the N pole is spinning in the same direction no matter the coil orientation). However, the Z coil is sensitive to the polarity simply because of the way this coil is built. Depending this coil is placed over its pads, on the one hand the current goes straight to the top of the coil and then starts spinning around while going down, or on the other hand, the current spins around while going up and then when reaching the top, the current goes straight to the bottom.
Hope the explanations are clear enough, feel free to ask for any further details.

Furthermore, we are following the North-East-Down requirements, which means the X axis is aligned along with the North, Y with East and the Z axis along with the ground.

You could check that out by simply orienting each and every coil along with the true North and check with the analyse widget whether you get the maximum amplitude or not : 
1 - Face the front side of the GPS toward North -> Make you get the maximum value
2 - Face the East side of the GPS toward the North -> Make you get the maximum value
3 - Face the Bottom side of the GPS toward the North -> Make you get the maximum value

Anyhow, if you believe you get a hardware error on this very unit, please get in touch with us by filling the form here : https://electronics.drotek.com/after-sale-service/

Paul

jinger26, 2021-02-10:
Is this still an open issue? @dagar @Paul-DROTEK ?