PX4/PX4-Autopilot #15313: qgroundcontrol error: Motor test command not supported

jimdgit, 2020-07-13:
**Describe the bug**
When you try and perform the motor test from the UI you get " Motor test command not supported"
**To Reproduce**
Click props not installed then move a slider.
You get the yellow error toast " Motor test command not supported'
**Expected behavior**
The motor turns on.

PX4 FMU V5
PX PRO v1.10.1 Stable Release

julianoes, 2020-07-14:
Motor testing is added with v1.11 as far as I know.

jimdgit, 2020-07-14:
Ok with 1.11 Beta 2 I get "motor test denied"

julianoes, 2020-07-15:
Cool, thanks for the feedback.

closed 2020-07-15 by julianoes

bkueng, 2020-07-16:
Just to clarify you did not arm, but just press the safety button, correct?

jimdgit, 2020-07-16:
Yes that’s exactly right just the safety button.

Sent from your iPhone