PX4/PX4-Autopilot #15317: GPIO Time based Camera Trigger not working in QGC mission

moonmotor, 2020-07-13:
Hi folks, I can not get the GPIO camera trigger to operate in the time based mode when running a mission in QGC. I checked previous trigger issues listed here, most seem to be from 2017,18 but could not find any related. Anybody else having this problem?

Very simple to setup/reproduce. Setup Camera for time based, on command and GPIO output. Set up the trig_ parameters for time, interval, active high, etc. Leave TRIG_PINS at default 56.

Planned a short mission with Camera trigger in Mission Start, also tried a Start image capture command, put the camera trigger command in a waypoint and tried every other combination of commands and settings, but can't get a trigger. Camera trigger test works, but after running a mission it sometimes gets locked out and we need to reboot. Time based, always on, works, but we can't have the camera continuously shooting pictures. Also, when we change mode to distance based, along with associated parameters, it works flawlessly.

It's only time time based, on command that will NOT work.

Any help would be appreciated,
Thanks,
MM

<width="775" alt="Screen Shot 2020-07-13 at 9 30 47 AM" src="https://user-images.githubusercontent.com/21185481/87310444-f7161680-c4eb-11ea-8fef-269348da7df3.png">

[image]

moonmotor, 2020-07-14:
I think I looked at every camera trigger issue in GitHub, PX4 and slack forums with no solution. If the camera can't be triggered with a GPIO output, it might just be good and come out and say that. The user guide is completely contradicting to other info on the sites. As an example I found this from a year ago, which makes NO sense at all. 
https://px4.slack.com/archives/C629RE9H7/p1562611063123900

If the camera is 0:Disable(ed), the user has no control and can't trigger the camera in a planned mission because the camera is DISABLED! So this is completely wrong. Am I the only one trying to trigger a camera in a mission on a periodic basis with the GPIO  #output?

'''
Don Gagne  12 months ago
If you are flying a camera mission should not be setting TRIG_MODE to anything other than 0:Disable. The mission commands should control everything. If you are flying a mission with camera commands and have TRIG_MODE also set to something else I have no idea what the interactions would be.

Don Gagne  12 months ago
If you are planning a Survey in QGC the camera trigger set is specified in the survey item itself. You are setting up camera triggering in Mission Start and then again in the Survey item. The 23.2... trigger distance is what is specified in the Survey item.

Don Gagne  12 months ago
The Survey item will also add a stop triggering command at the end of the Survey which is what you are seeing.
'''

hamishwillee, 2020-07-14:
Yes, the camera docs are clearly not correct - the QGC screen is different than current version and the http://docs.px4.io/master/en/peripherals/camera.html#trigger_mode isn't qutie the same as http://docs.px4.io/master/en/advanced_config/parameter_reference.html#TRIG_MODE

@moonmotor There are quite a lot of conditions you have mentioned. Let's just start with what you want to do. I THINK this is time based commands triggered by a mission command (`MAV_CMD_DO_SET_CAM_TRIGG_DIST`) and stopped by that same command at the end. Is that correct.

@bkueng According to the docs this should work with TRIG_MODE = 4. Is that right? I'm not sure what Don was referring to above by saying the mode should be zero. I think that means "off" too.

bkueng, 2020-07-14:
Yes I would expect so. `TRIG_MODE=1 Time based, on command` should start/stop triggering on `VEHICLE_CMD_DO_TRIGGER_CONTROL`. Is that command sent?

moonmotor, 2020-07-14:
Thanks for the quick replies guys! I'm trying to trigger the camera on a time basis. We could write code to turn it on before a mission and then off after the mission, but that's cumbersome. We'd like to use the mission commands to turn the camera on/of during our missions. The most obvious setting is TRIG_MODE=1 (time based, on command), which doesn't work. I did see in the documentation somewhere that said use TRIG_MODE=4 and we tried that with no luck. 
@bkuend, we are setting camera to start in the Mission Start tab, so it doesn't say what command is being sent. If we send a command explicitly it sends the MAV_CMD_IMAGE_START_CAPTURE. But we have also included the Trigger Control command MAV_CMD_DO_TRIGGER_CONTROL and didn't have and luck with that either. It seems like such a basic function, but we have been working on it for days. Your help is very much appreciated!

moonmotor, 2020-07-14:
Another bit of information: It did randomly work twice over the past three days, but was unrepeatable - so that gave us hope we were doing something wrong. Second thing is that in many cases, after running a mission the camera will no longer trigger in the Camera Test screen above. That's why we put in the MAV_CMD_DO_TRIGGER_CONTROL command thinking the camera might need to be (re)enabled first? And again we're trying to use the GPIO output to close a relaying to trigger a "dumb" camera - no PWM communications.
Thanks again!

moonmotor, 2020-07-14:
Sorry to bombard with messages, but we need to solve (what should be) a trivial problem and get out imaging crops. Your help continues to be greatly appreciated. Here's a screen capture from QGC - since we're working within the confines of QGC  - that's where we need to solve the problem.
[image]

This happens to be mode 4, but we also tried mode 1 (0,2,&3).
We can trigger from the Mavlink console, start the trigger (which snaps a photo), stop the trigger, but not able to get it running on the time-based mode.
Losing hair, sleep and customers!

hamishwillee, 2020-07-15:
@moonmotor I heartily agree this should be a trivial problem and fix. But this is an open source project, so while we will try help, unless a problem affects vehicle safety it is no one's urgent responsibility. You may need to dig into the code yourself or invest in consultancy if we can't help. I have never tried this because I don't have the hardware. 

The [code is here](https://github.com/PX4/Firmware/blob/master/src/drivers/camera_trigger/camera_trigger.cpp). 

Some observations.
1. MAV_CMD_IMAGE_START_CAPTURE and friends are a newer mavlink interface than the other message. These are generally supported by MAVLink cameras but do not appear to be supported by this driver (and should be). 
1. The driver also supports [MAV_CMD_DO_SET_CAM_TRIGG_INTERVAL](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_CAM_TRIGG_INTERVAL) (not documented in camera trigger docs)  - @bkueng correct?

My interpretation of the docs is that time triggering **in survey missions** is not supported - only distance triggering is supported. So what "should work" is that you set `TRIG_MODE=4` and the correct pins etc. Then QGC should automatically set `MAV_CMD_DO_SET_CAM_TRIGG_DIST` in the survey mission. If it does not do that then that's possibly a bug in QGC.

So question, are you doing a survey mission, or some other mission? If it is a survey mission, are those triggers being set - if not, what is being set instead?

It might be that MAV_CMD_IMAGE_START_CAPTURE is being set, in which case perhaps the easiest fix would be to PR in support for that message.

@bkueng Anything else you can infer from this/corrections to my comment?

bkueng, 2020-07-15:
@moonmotor your screenshot suggests `camera_trigger` was not running and you had to start it, is that correct? Which firmware version is that?

> The driver also supports MAV_CMD_DO_SET_CAM_TRIGG_INTERVAL (not documented in camera trigger docs) - @bkueng correct?

Correct.

To further debug, I suggest setting `TRIG_MODE=1`, then sending `VEHICLE_CMD_DO_TRIGGER_CONTROL` with `param1=1` and `param3=0`, and then look at `camera_trigger status`.

moonmotor, 2020-07-15:
@bkueng that's the secret code!
MAV_CMD_DO_TRIGGER_CONTROL 1,0,0 starts the camera triggering based on the parameters setup in Camera Trigger with TRIG_MODE=1 . None of the other camera commands in QGC Mission Plan seem to have any effect.
MAV_CMD_DO_TRIGGER_CONTROL 0,0,0 stops the triggering.
It's not quite reliable - sometimes the camera will start triggering as soon as the mission is uploaded, and often times the camera will start triggering again after the aircraft has landed.
Before, we couldn't get the camera to start triggering, now we can't get it to stop. hahaha but we're getting much closer!
Thanks!

moonmotor, 2020-07-16:
@bkueng - is there another way to disable the camera in MAVSDK-Python? If we cold put an additional command in our script to turn the camera trigger off (a second time) after the completion of a mission, that would be a temporary work around. We could make the modification(s) in cpp, but that would be a more involved development and right now our time is short.
MM

bkueng, 2020-07-16:
Good!

@julianoes knows if you can send arbitrary commands from python MAVSDK.

> It's not quite reliable - sometimes the camera will start triggering as soon as the mission is uploaded, and often times the camera will start triggering again after the aircraft has landed.

I remember there was a bug in navigator that would do that if the first mission item was a command. Which version are you on? If it's that then a work-around is to add another item before the command.

julianoes, 2020-07-16:
No you can't send arbitrary commands from MAVSDK-Python. MAVLink passthrough is only available for C++ so far.

In theory there should be some logic in navigator to stop triggering after the mission is done.

Or would this help?
https://github.com/PX4/Firmware/pull/13983/files

moonmotor, 2020-07-16:
I didn't think we could send a command through Python and didn't want to get into the C++ code at this time.
We're running the latest stable build of PX4 and QGC 4.0.9.
@bkueng what would a non-command be? It seems like all of the actions are commands. The first line of the mission is always Mission Start. I don't think we can put anything in before that line.

moonmotor, 2020-07-16:
SOLVED: sort of - we'll be working on a cleaner solution.
We put a takeoff command at the beginning of the mission and that seems to eliminate the camera triggering again after the mission is complete. So, for anyone wanting to trigger a camera using the basic GPIO (relay) output, the workflow is something like this: Setup the camera and Camera Trigger parameters as shown above. Use MAV_CMD_DO_TRIGGER_CONTROL to turn the triggering on or off parameter 1=1 =>on, parameter 1=0=> off. be sure to use a takeoff command as shown below.
[image]