PX4/PX4-Autopilot #15328: make px4_sitl jmavsim                  error

yun-max, 2020-07-14:
~/src/Firmware$ make px4_sitl jmavsim
[2/738] git submodule src/lib/ecl
[7/738] git submodule src/drivers/gps/devices
[11/738] git submodule mavlink/include/mavlink/v2.0
[216/738] git submodule Tools/jMAVSim
[685/738] Building CXX object src/syst...tests.dir/test_microbench_atomic.cpp.o
FAILED: /usr/bin/ccache /usr/bin/c++   -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"tests\" -DPX4_BOARD_NAME=\"PX4_SITL\" -DPX4_MAIN=tests_app_main -D__CUSTOM_FILE_IO__ -D__PX4_LINUX -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I../../boards/px4/sitl/src -I../../platforms/posix/src/px4/common/include -I. -Isrc/lib -I../../platforms/posix/src/px4/generic/generic/include -I../../platforms/common/include -I../../src -I../../src/include -I../../src/lib -I../../src/lib/matrix -I../../src/modules -I../../platforms/posix/include -Iexternal/Install/include -I../../src/lib/ecl -O2 -g -DNDEBUG -fPIC   -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fdiagnostics-color=always -fno-builtin-printf -fno-strength-reduce -Wformat=1 -Wunused-but-set-variable -Wno-format-truncation -fcheck-new -fno-exceptions -fno-rtti -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -Wno-double-promotion -Wno-float-equal -Wno-missing-declarations -Wno-sign-compare -Wno-unknown-warning-option -Wno-unused-but-set-variable -Wno-unused-result -Wno-unused-variable -Wno-vla-larger-than -std=gnu++14 -MMD -MT src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_microbench_atomic.cpp.o -MF src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_microbench_atomic.cpp.o.d -o src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_microbench_atomic.cpp.o -c ../../src/systemcmds/tests/test_microbench_atomic.cpp
In file included from ../../src/systemcmds/tests/test_microbench_atomic.cpp:49:0:
../../platforms/common/include/px4_platform_common/atomic.h: In instantiation of ‘bool px4::atomic<T>::compare_exchange(T*, T) [with T = bool]’:
../../src/systemcmds/tests/test_microbench_atomic.cpp:184:2:   required from here
../../platforms/common/include/px4_platform_common/atomic.h:169:34: error: parameter ‘expected’ set but not used [-Werror=unused-but-set-parameter]
  inline bool compare_exchange(T *expected, T desired)
                                  ^
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[685/738] Building CXX object src/syst...ystemcmds__tests.dir/test_matrix.cpp.o
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

xdwgood, 2020-07-15:
Please try the latest master program, the problem has just been solved

julianoes, 2020-07-15:
Fixed in https://github.com/PX4/Firmware/pull/15323.

closed 2020-07-15 by julianoes