PX4/PX4-Autopilot #15336: LIDAR-Lite v3 pwm doesn't work

subakioinc, 2020-07-15:
**Describe the bug**
Flight Controller : Pixhawk2.1

On master version :
 LIDAR-Lite v3 pwm doesn't work.

Current Stable 1.10.1 version : 
 LIDAR-Lite v3 pwm works.

Parameter setting for master and stable version  : 
 * SENS_EN_LL40LS : PWM
 * EKF2_HGT_MODE : Range sensor
 * EKF2_RNG_AID : Range aid enabled

Wiring : 
 * Same wiring for master and stable version.

**Expected behavior**
 LIDAR-Lite v3 pwm works.

**Log Files and Screenshots**
[image]
PWM works on Stable 1.10.1 version. 

----

Below on Master version. PWM doesn't work. 
[image]
[image]

julianoes, 2020-07-15:
Let me ping @mcsauder

dagar, 2020-07-15:
Could you try current master or PX4 v1.11 beta? Since v1.10 the pwm_input driver has been updated and the lidar lite PWM has been split out as a standalone driver (https://github.com/PX4/Firmware/tree/master/src/drivers/distance_sensor/ll40ls_pwm).

subakioinc, 2020-07-15:
I tried 1.11.0-beta2 and current master. Both of them have same error. 
I checked previous ll40ls module is split into 2 modules(ll40ls(for I2c) and ll40ls_pwm).

Error on 1.11.0-beta2 :
[image]
 
-----

OK 1.10.1 stable :
[image]

JulianSchliessus, 2024-03-15:
Still having the same issue with versions above 1.10.1 when using ll40ls with pwm

julianoes, 2024-03-18:
> Still having the same issue with versions above 1.10.1 when using ll40ls with pwm

Can you try with v1.14? v1.10 is ancient.

JulianSchliessus, 2024-03-18:
I already tried 1.14.0, it just seems that beginning from 1.10.2 there are some changes which lead to this issue.

JulianSchliessus, 2024-03-18:
> > Still having the same issue with versions above 1.10.1 when using ll40ls with pwm
> 
> Can you try with v1.14? v1.10 is ancient.

If you need any details to verify this issue, don't hesitate to ask :smile: 
My wild guess for the problem source was the pwm_input driver, which does not detect a signal after the update to the new version.

julianoes, 2024-03-18:
Oh "above that version", sorry I understand you now.

JulianSchliessus, 2024-03-18:
Yes I can still reproduce the error mentioned previously in this issue, also with newer versions of PX4.
I’m building on px4_fmu-v5_default.

When running `ll40ls_pwm status` in the MAVLink Console, I'm getting the following response with firmware v1.14.0 and v1.10.2:
```shell
ll40ls_pwm status
ll40ls: read: 1941 events, 3301us elapsed, 1.70us avg, min 1us max 35us 2.027us rms
ll40ls: comms errors: 1940 events
ll40ls: resets: 0 events
ll40ls: zero resets: 1 events
poll interval:  50000
```
so the module is mainly throwing comm errors.

When running the command `ll40ls status` on v1.10.1, I'm getting a valid response:
```shell
ll40ls status
ll40ls: read: 3468 events, 93625us elapsed, 27.00us avg, min 20us max 232us 16.026us rms
ll40ls: comms errors: 533 events
ll40ls: resets: 0 events
ll40ls: zero resets: 0 events
poll interval:  50000
```
There are still some comm errors but a big part of the data gets through ...

JulianSchliessus, 2024-04-01:
So after some tiral and error it seems that the wiring of the lidar was wrong.
The sensor was already wired for an old project which worked with the old firmware versions.
Looking in the current [documentation of the lidar lite](https://docs.px4.io/main/en/sensor/lidar_lite.html), it seems that the wiring has changed.
Maybe it should be pointed out in the documentation, that the wiring is only compatible with specific firmware versions.

julianoes, 2024-04-01:
Oh, that's tricky. What changed in the wiring?

JulianSchliessus, 2024-04-01:
So the wiring which apparently works with the old firmware versions was set up by someone in the past, who is not in reach anymore :grin:
[image]
There is a resistor shortening ground and one line which is going into PWM.

In the new wiring setup from the [documentation](https://docs.px4.io/main/en/sensor/lidar_lite.html), there is no shortening of two ports through a resistor (but only a resistor on one port):
[image]

I'm sorry to provide only this bad picture but I do not have better pictures for instance, as the sensor is built in the plane and the cables are quite hidden.
Maybe there are some old wiring plans from the older firmware versions?

DARKMOONlite, 2024-12-04:
I spent a little bit of time trying to diagnose what was going wrong, trying to get this working on a pixhawk 1 2MB before I stumbled upon this treasure trove of information. I thought I might summarise my findings for all newbies like me working with px4.
1. Ensure that the `LL40LS_PWM` firmware is installed, this can be done by building from [source](https://docs.px4.io/main/en/dev_setup/building_px4.html) and ensuring that the [Driver Kconfig file](https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/distance_sensor/Kconfig) contains the line `select DRIVERS_DISTANCE_SENSOR_LL40LS_PWM` this should allow you to run via a mavlink terminal such as the one in QGC: `ll40ls_pwm status` which should report the status of the driver.
- *note: make sure you select a specific version of px4 when cloning the repo, i.e. `v1.15.2`, otherwise you may get a alpha/prerelease version that for me... didn't work.*
2. Wiring: to wire follow the above image, connecting the red 5V wire to any 5V pin on the pwm out pin, the black Gnd wire to any Ground pin, and the yellow mode select pin 
to the pixhawk's pin **55**, and finally connect a **470 ohm** resister between the yellow wire (pin 55) and pin **54** I've also added a simple diagram hopefully showing how to wire it up with a pixhawk 1 example.
[image]

julianoes, 2024-12-04:
Thanks @DARKMOONlite! And you got it working? In that case we can close the issue.

DARKMOONlite, 2024-12-05:
@julianoes yes it is perfectly, thank you and others that contributed to this thread. I would say its probably safe to close the issue.

closed 2024-12-09 by julianoes

Seung-W, 2024-12-19:
@julianoes @DARKMOONlite I tried your wiring on Pixhawk 6x and v1.15.2. However it does not work. Did you tried that wiring for latest version pixhawk? Also, I wonder how you figured it out.

I wanted the `ll40ls_pwm` driver to start on PX4 boot, so I added `select DRIVERS_DISTANCE_SENSOR_LL40LS_PWM` to Distance sensor Kconfig file and add `CONFIG_DRIVERS_PWM_INPUT=y` to ~/PX4-Autopilot/boards/px4/fmu-v6x/default.px4board file. This ensured that the `ll40ls_pwm` driver was running when the PX4 booted.

However, `DISTANCE_SENSOR` messages still does not appeared at MAVLink Inspector. I also tired wiring from [documentation](https://docs.px4.io/main/en/sensor/rangefinders.html), it does not appeared.

[image]

DARKMOONlite, 2024-12-21:
Hey @Seung-W, I haven't tried it on a pixhawk 6 yet, however I have bought a very similar model I'll be testing in the new year, once I'm back from holiday.

But based on what you've written it sounds like a power or wiring issue,  how are you powering the px4? I found that if you just power it via the USB connection it doesn't actually power the pwm pins as the max of USB is ~1A when the pixhawk expects ~2.5A.

You could also try using different pins, the enable pin is defined by the [ll40ls_pwm.h](https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/distance_sensor/ll40ls_pwm/LidarLitePWM.h) file.
I started looking into using the firmware with different pin combinations by modifying the CPP files, but stopped as I had other more pressing matters. Maybe keep that idea in the back pocket as it does have its issues.
Sorry I can't help more for the time being, lmk if there's anything else I can help with

Seung-W, 2024-12-29:
Happy New Year @DARKMOONlite ! First of all, thank you for your reply. I checked and found that when I supply power to the POWER port, there is no voltage applied to the AUX port. So I made a separate wire from the battery and fed it to the AUX port separately, and the lidar works fine with the lidar wiring you provided. Thank you!