PX4/PX4-Autopilot #15337: mavlink stream consistency in SITL with 1.10.2

asherikov, 2020-07-15:
It appears that data can be rather inconsistent between different mavlink streams, for example, 'landed state' flag reported in extended state differs in 'normal' (GCS) and 'onboard' mavlink streams (data is printed from `MavlinkStreamExtendedSysState::send()`):
```
1594807716 landed_state: 1 | mavlink type: Onboard
1594807716 landed_state: 1 | mavlink type: Onboard
1594807716 landed_state: 1 | mavlink type: Normal
1594807716 landed_state: 1 | mavlink type: Onboard
1594807716 landed_state: 3 | mavlink type: Onboard
1594807717 landed_state: 3 | mavlink type: Onboard
1594807717 landed_state: 3 | mavlink type: Onboard
1594807717 landed_state: 3 | mavlink type: Onboard
1594807717 landed_state: 1 | mavlink type: Normal
1594807717 landed_state: 3 | mavlink type: Onboard
1594807717 landed_state: 3 | mavlink type: Onboard
1594807718 landed_state: 3 | mavlink type: Onboard
1594807718 landed_state: 3 | mavlink type: Onboard
1594807718 landed_state: 1 | mavlink type: Normal
1594807718 landed_state: 3 | mavlink type: Onboard
1594807718 landed_state: 3 | mavlink type: Onboard
1594807718 landed_state: 3 | mavlink type: Onboard
1594807719 landed_state: 3 | mavlink type: Onboard
1594807719 landed_state: 3 | mavlink type: Onboard
1594807719 landed_state: 3 | mavlink type: Onboard
1594807719 landed_state: 1 | mavlink type: Normal
1594807719 landed_state: 3 | mavlink type: Onboard
1594807719 landed_state: 3 | mavlink type: Onboard
1594807720 landed_state: 3 | mavlink type: Onboard
1594807720 landed_state: 3 | mavlink type: Onboard
1594807720 landed_state: 1 | mavlink type: Normal
1594807720 landed_state: 3 | mavlink type: Onboard
1594807720 landed_state: 2 | mavlink type: Onboard
1594807720 landed_state: 2 | mavlink type: Onboard
1594807721 landed_state: 2 | mavlink type: Onboard
1594807721 landed_state: 2 | mavlink type: Onboard
1594807721 landed_state: 1 | mavlink type: Normal
1594807721 landed_state: 2 | mavlink type: Onboard
1594807721 landed_state: 2 | mavlink type: Onboard
1594807721 landed_state: 2 | mavlink type: Onboard
1594807722 landed_state: 2 | mavlink type: Onboard
1594807722 landed_state: 2 | mavlink type: Onboard
1594807722 landed_state: 2 | mavlink type: Onboard
1594807722 landed_state: 1 | mavlink type: Normal
1594807722 landed_state: 2 | mavlink type: Onboard
1594807722 landed_state: 2 | mavlink type: Onboard
1594807723 landed_state: 2 | mavlink type: Onboard
1594807723 landed_state: 2 | mavlink type: Onboard
1594807723 landed_state: 2 | mavlink type: Onboard
1594807723 landed_state: 1 | mavlink type: Normal
1594807723 landed_state: 2 | mavlink type: Onboard
1594807723 landed_state: 2 | mavlink type: Onboard
1594807724 landed_state: 2 | mavlink type: Onboard
1594807724 landed_state: 2 | mavlink type: Onboard
1594807724 landed_state: 1 | mavlink type: Normal
1594807724 landed_state: 2 | mavlink type: Onboard
1594807724 landed_state: 2 | mavlink type: Onboard
1594807724 landed_state: 2 | mavlink type: Onboard
1594807725 landed_state: 2 | mavlink type: Onboard
1594807725 landed_state: 2 | mavlink type: Onboard
1594807725 landed_state: 2 | mavlink type: Normal
```
landed state flag values:
```
   MAV_LANDED_STATE_UNDEFINED=0, /* MAV landed state is unknown | */
   MAV_LANDED_STATE_ON_GROUND=1, /* MAV is landed (on ground) | */                                                    
   MAV_LANDED_STATE_IN_AIR=2, /* MAV is in air | */
   MAV_LANDED_STATE_TAKEOFF=3, /* MAV currently taking off | */
   MAV_LANDED_STATE_LANDING=4, /* MAV currently landing | */
   MAV_LANDED_STATE_ENUM_END=5, /*  | */
```
Note that in this case 'takeoff' state does not appear in 'normal' stream at all.

The issue is not persistent. Is this expected behavior or a bug?

julianoes, 2020-07-15:
Thanks for the issue @asherikov. That's certainly odd but doesn't surprise me as we've seen weird effects in MAVSDK with landed_state before.

Have you checked if this is fixed with v1.11?

asherikov, 2020-07-15:
I haven't tried 1.11, I'll try to find time to check.

dagar, 2020-07-17:
Yes please let us know if this is an issue v1.11 beta or current master. Quite a lot has changed in mavlink this year with moving to the new uORB::Subscription API.

asherikov, 2020-07-24:
1. I am having issues with AirSim SITL with newer firmware, sorry I don't want to dive into this atm.
2. In my case changing the mavlink stream mode to onboard resolved the issue, or at least made inconsistencies negligible.