PX4/PX4-Autopilot #15340: uorb top not showing uORB message, even though it is streaming

js-motocrane, 2020-07-15:
**Describe the bug**
I followed the steps in the PX4 documentation to create a custom uORB message and publish it, within my custom application. I went to the nsh shell in QGroundControl to try and confirm my message was streaming by typing `uorb top` , however when I did this, I did not see my custom uORB topic anywhere in the list (I thoroughly searched through it). Based on a suggestion from the PX4 Slack message I posted, I tried `uorb top my_uorb` and it displayed the topic there correctly. 

Worth noting is that once I got this uORB message to feed a custom MAVLink message, it is displayed in the `uorb top` list. However, when this is not happening (and when there are no other subscribers to that uORB topic), the uORB message is not displayed in the list. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create new uORB message, advertise and publish it from a running application/module
2. Use the `uorb top` command in the nsh shell (from QGC)

**Expected behavior**
The custom uORB message should be there in the list of topics.

**Log Files and Screenshots**
Sorry, I did not capture any screenshots, but I hope the post is self-explanatory.

**Drone (please complete the following information):**
-Using Pixhawk Cube Black as hardware running this code

**Additional context**
None

dagar, 2020-07-15:
To try and keep the full `uorb top` output to a reasonable length by default only topics with subscribers are printed. 

https://github.com/PX4/Firmware/blob/cfcdab39929bd29c292082292b1c0824c0b75e29/src/modules/uORB/uORBDeviceMaster.cpp#L412

js-motocrane, 2020-07-15:
Totally makes sense. I should have spent more time looking through the code. No issue then. I will close this.

closed 2020-07-15 by js-motocrane