PX4/PX4-Autopilot #15344: 360 Degree Collision Prevention

KarlEaves, 2020-07-15:
Hey PX4 Team!

I was wondering if there's any plug-and-play options for using multi-directional distance sensors. I found this multi-directional distance sensor that uses 4/8 TeraRanger Evo 60m (PX4 Compatible Range Finders) https://www.terabee.com/shop/lidar-tof-multi-directional-arrays/teraranger-tower-evo/ and was hoping to use this to perform 360 degrees collision prevention

If this isn't an option, I was wondering if multiple sensors could be plugged into a companion computer running https://github.com/PX4/avoidance ? Or if we could configure multiple sensors without using the PX4/avoidance repo, since all we really need is Collision Prevention and not Obstacle Avoidance?

 I am confused on how the companion computer is supposed to send OBSTACLE_DISTANCE messages for multiple sensors to PX4, and if PX4 is able to interpret these messages. Is PX4 able to get sensor data from 4 cameras facing N,E,S,W and merge that data to perform collision prevention for all sides?

hamishwillee, 2020-07-15:
@KarlEaves 

> I am confused on how the companion computer is supposed to send OBSTACLE_DISTANCE messages for multiple sensors to PX4, and if PX4 is able to interpret these messages. Is PX4 able to get sensor data from 4 cameras facing N,E,S,W and merge that data to perform collision prevention for all sides?

The answer to this part is here: http://docs.px4.io/master/en/computer_vision/collision_prevention.html#algorithm

Essentially the message needs to map the data from sensors around the vehicle into 36 sectors.

There is no plug'n'play solution for plugging such a sensor into the flight stack - wouldn't it be nice if the sensor had a mavlink interface that produced the data as we require :-). the plug'n'play solution currently only supports one sensor in one direction (as per link above).

Yes it should be possible for a collision prevention system to do away with PX4/avoidance and have a simplified mechanism for producing the OBSTACLE_DISTANCE on a companion, but I don't know of such work.

FYI @jkflying

KarlEaves, 2020-07-21:
@hamishwillee  Thanks for the info!  I will try creating the program on the companion to send OBSTACLE_DISTANCE data. I am currently running v1.8, but it appears this also supports collision prevention from a companion.

Thank you for the help!

closed 2020-07-21 by KarlEaves