PX4/PX4-Autopilot #15347: Drone Crash: Motors shutdown on fly

FlavioTonelli, 2020-07-16:
During some flight test we had an hexarotor crash. All the motors stopped suddenly and the machine crashed on the ground.
We right now can't explain such a behavior.
Here is the flight log: https://review.px4.io/plot_app?log=f177f3be-1ffc-4040-a86d-5831aefa752c
Can you guys spot any SW related issue?
The firmware is based on bcce75e691657e7da06ff1bf88fc878eb026c92b. It's an April master commit.
The modification we made to the firmware are minors and do not related to navigation. I do not think they can explain such a think.

kuiskas, 2020-07-16:
Hi @FlavioTonelli ,
I am sorry to heard that. I had the same issue a week ago, but I was using the stable 1.10.1.  https://github.com/PX4/Firmware/issues/15289
I have now not much time to look at the log, but I will do as soon as I can. 
What I did see is, you are using LNDMC_Z_VEL_MAX, 0.5. The lowest hard limit is 0.6, the default 0.7. At least it was in the 1.10.1.
I do not know how this could happend. You can take a look here https://github.com/PX4/Firmware/issues/13124

FlavioTonelli, 2020-07-16:
Hello @kuiskas, thank you for your reply.
I'm not expecting LNDMC_Z_VEL_MAX to be out of the default range because I never force saved that. Do you think that parameter can be involved in the crash? In that case the px4log shall show some landed related message, doesn't it?

dagar, 2020-07-16:
The outputs do stop before the land detector triggers.

[image]

One possible culprit I'm investigating is the safety button, but the minimal logging rate (1 Hz) makes it hard to see if it's cause or effect.

[image]

dagar, 2020-07-16:
As I go through this I'm adding logging data I wished we had. https://github.com/PX4/Firmware/pull/15355

dagar, 2020-07-16:
@FlavioTonelli can you comment on the safety button? Is it part of the GPS puck or separately wired? Which model is it?

FlavioTonelli, 2020-07-16:
Hello @dagar the safety button is part of the gps antenna. The model we use it the Hex Here2 antenna. Could it be an electrical false contact on that?

kuiskas, 2020-07-16:
it did disarm before landing detection.
[image]
Same happened to me. Is this parameter LNDMC_Z_VEL_MAX not relevant anymore?

kuiskas, 2020-07-16:
[image]
I believe it was the safety switch. could be an explanation? Is this hardware or could be software related? Should not be in the software an impediment to disarm if we are in the air?

kuiskas, 2020-07-17:
Hi again, 
I just could  reproduce the problem again, this time on our test stand, so no hardware damage. The test procedure was:
1. Octocopter fixed on our test stand. Armed and disarmed several times. 
2. At the end I armed the octocopter and while hovering, pressed the safety switch. Copter disarmed before the Maybe landing state, see picture.
[image]

I was using version 1.10.1, with slightly modifications. As @FlavioTonelli  the Here 2 RTK GPS...

Here is the log from this test. 
https://logs.px4.io/plot_app?log=e8ccb679-491e-4670-9bfb-7099a7bf0f87

I am starting to be pretty sure, the problem comes from the GPS safety button. Can someone confirm it? @dagar 
Also I believe the copter should not be able to disarm once it is in the air. The concept of a safety switch is to not arm, when undesired, but it should not be able to disarm a copter. Normally one armed, the user has no possibility to reach this button without risking the arms with the propellers. What do you think?

dagar, 2020-07-17:
> I am starting to be pretty sure, the problem comes from the GPS safety button.

Likewise, although I don't think we have the data to prove it definitively this time.

I agree with the idea of it not being able to disarm (at least by default) and I've started looking at making the change.

kuiskas, 2020-07-17:
@FlavioTonelli which flight control were you using?.
@dagar I agree that we cannot absolutely know for sure if this is 100% the problem. Anyway as the same issue happened to me, to @FlavioTonelli and I could recreate the same (similar) log output on a test stand. I would say that there are too many coincidences.
Talking to colleagues we realize some time ago, we tried to connect the pixhawk 4 GPS to the black cube. Safety switch was not recognized. Here 2 with pixhawk 4 normally works, but Iam starting to thing that they work electrically different and this could be a lead to the problem, if this is the problem. I have right now no possibility to test it. I will do as soon as I can. Maybe you know better.

FlavioTonelli, 2020-07-18:
@kuiskas we are using the Pixhawk4 board. I do not think the main problem is an electrical compatibility between the Here2 and the board. We made a lot of hours of flight with the same hardware configuration on different machines.
Right now I strongly think the problem was with that single antenna. Doing some inspection on it, the plastic button it's slightly damaged. That could explain the on fly shutdown. The button could anyway even being damaged after the crash. So I can not be totally sure.
@dagar 
> makes it hard to see if it's cause or effect.
How can the safety switch state change in the log as an effect? That state shall be strongly independent from the other px4 modules,isn't it? What I'm stating is that is really hard let that state change without a real button pressure. What's your opinion on that?

kuiskas, 2020-07-20:
Well @FlavioTonelli  same hardware here, same issue here. We had also lot of hours in the air before the crash. If I am not wrong, from what I remember, PX4 polls the safety switch at 10 Hz, 10 times. So in case of a malfunctioning button it is not easy that this happens, but not impossible and I believe we both has the same problem. Let see if the team con confirm this and make some changes on the firmware.

mehmetyldz87, 2020-07-24:
I had the same situation months ago. 
Forum topic is here. 
https://discuss.px4.io/t/firmware-1-10-1-stable-release-motor-disarm-issue-in-flight/14858/9
Thank you

FlavioTonelli, 2020-07-24:
Hello @mehmetyldz87, after a quick read to the discussion topic I think the issue are unrelated.
In your case, the rc itself accidentally trigger the arm switch. When in manual mode the arm switch is able to disarm the drone on fly. On my point of you this is not a bug or a SW architectural problem.
Our discussion is focused on the safety switch, that shall be not able to disarm on fly.

mehmetyldz87, 2020-07-24:
Hello @FlavioTonelli  

It doesn’t seem to be an RC glitch issue or arm/disarm issue. I am not sure about the problem. Therefore I am sharing under this issue.

kuiskas, 2020-09-10:
Hello,
is this actually still open? I have seen this changes on the commander.cpp to prevent disarm flight.
[image]
https://github.com/PX4/Firmware/blob/9116acc431b64ad7d824ec5a71f3dc7acbcb1912/src/modules/commander/Commander.cpp#L1546
It is solved?

FlavioTonelli, 2020-09-10:
Looks to me that the modification you highlighted is related to https://github.com/PX4/Firmware/pull/13992/commits/f72ed580c82ab8b1919e5fa102271bbbaea5af28.

That was merged before my crash.

LorenzMeier, 2021-02-03:
We will change the safety button to be sticky (you will not be able to disable safety in the future via the button). The logic that is already present should cover the expected cases.

closed 2021-02-03 by LorenzMeier