PX4/PX4-Autopilot #15361: Run PX4 on crazyflie2.1

LJ0707, 2020-07-17:
I want to run PX4 on crazyflie2.1.But now PX4 only support crazyflie2.0.I know there are some difference between crazyflie2.0 and crazyflie2.1.
* crazyflie2.0:  https://store.bitcraze.io/products/crazyflie-2-0
* crazyflie2.1:  https://store.bitcraze.io/products/crazyflie-2-1
So I add bmi088 and bmp388 drivers in ./boards/bitcraze/crazyflie/default.cmake.
~~~
DRIVERS
		barometer/lps25h
		barometer/bmp388
		distance_sensor/vl53lxx
		gps
		imu/bmi088
		imu/mpu9250
		optical_flow/pmw3901
		px4fmu
~~~
And I start them in ./boards/bitcraze/crazyflie/init/rc.borad_sensors.
~~~
adc start

# Onboard I2C
mpu9250 -I -R 12 start
#bmi088 -s -A -G -R start
bmi088 -A -R 10 -s start

bmi088 -G -R 10 -s start

# I2C bypass of mpu
lps25h -I start
bmp388 -I start

# Optical flow deck
vl53lxx start -X
~~~
But in qgroundcontrol when I input command to check sensors status，there is no sensor.
[image]
[image]

dagar, 2020-07-17:
Waiting on I2C support for the BMI088.

datwwe, 2020-08-04:
> I want to run PX4 on crazyflie2.1.But now PX4 only support crazyflie2.0.I know there are some difference between crazyflie2.0 and crazyflie2.1.
>
>     * crazyflie2.0: [...]

Hi, i'm trying to upload px4 firmware onto CF2.1 and you seems to made it. Can you please help?

LJ0707, 2020-08-07:
1. make bitcraze_crazyflie_default menuconfig
Go to: Devices Drivers -> Serial Driver Support -> Serial Console . And select the USART3.
2.make bitcraze_crazyflie_default upload. 
I follow this intruction to upload px4 firmware onto CF2.1.
https://docs.px4.io/v1.9.0/en/complete_vehicles/crazyflie2.html

But when I upload px4 firmware onto CF2.1.I cannot start sensors.Can you please help.

dagar, 2020-08-07:
Can you try this PR? https://github.com/PX4/Firmware/pull/15421

datwwe, 2020-08-14:
> Can you try this PR? #15421

The CF2.1 use BMI088(not mpu9250 like CF2.0) imu so i think we also needs some drivers( like other imu in src/drivers/imu) not just editting some configs..
Btw, are you developing on CF2.1?

LJ0707, 2020-08-14:
I find a  bmi088 driver.It support i2c.
https://github.com/BoschSensortec/BMI08x-Sensor-API
Now I am trying to add it to CF2.1.But I am not sure it will work.
Could you give me some idea?

datwwe, 2020-08-14:
> I find a bmi088 driver.It support i2c.
> https://github.com/BoschSensortec/BMI08x-Sensor-API
> Now I am trying to add it to CF2.1.But I am not sure it will work.
> Could you give me some idea?

if you direct to /src/drivers/imu/borsch/ you'll see BMI088, and BMP388 is in /src/drivers/barometer.
So i think u should edit the config to those direction.
P/s : can i add your contact for more instance reply cause i'm also making some stuffs wwith CF2.1 PX4

LJ0707, 2020-08-14:
Thank you.My mail address is 2276835336@qq.com.

KimHyungSub, 2020-08-27:
@LJ0707 @datwwe 
Are you guys having any progress? I am also working with CF2.1. But, I have the same problem.

diegomrt, 2020-09-24:
Hi @KimHyungSub @LJ0707 @datwwe,

Any progress so far? It would be nice to know if you finally made PX4 work on our Crazyflies 2.1

jgoppert, 2020-09-30:
@dagar  @LJ0707  We are starting to work on this as well for crazyflie 2.1 support. We have tried https://github.com/PX4/Firmware/pull/15552, but it doesn't seem like that was taken very far except for the bringing in the bosch code, it doesn't seem to build the module.

dagar, 2020-09-30:
I have a Crazyflie v2.1 in a box somewhere, I'll take a look this weekend.

dagar, 2020-10-05:
I'm working on adding generic register read/write helpers to the I2C/SPI interfaces so we can cleanly support drivers like the BMI088 for both.

jgoppert, 2020-10-05:
Thanks, really appreciate it. Let me know if there is anything we can do to help.

jgoppert, 2020-10-07:
@dagar Do you have a branch going that we could help test and possibly help with development?

jgoppert, 2020-10-14:
@dagar Any updates on this? We would really like to push this through to support PX4 on the crazyflies in our lab. We are willing to do the leg work, but just want to make sure we sync up with you. If you can get us started, we can do the rest.

zp-yang, 2020-10-28:
While trying to connect bmi088 on SPI bus 
`initSPIDevice(DRV_ACC_DEVTYPE_BMI088, SPI::CS{GPIO::PortB, GPIO::Pin1}),
initSPIDevice(DRV_GYR_DEVTYPE_BMI088, SPI::CS{GPIO::PortB, GPIO::Pin0}),`
The probe() function gives a erroneous reading of chip_id.

From gdb, the 'ACC_CHIP_ID' read is '0xff', while the actual 'ID' is '0x1e'.

Any ideas why this happens? Thanks! @dagar @LJ0707

JizhouChen, 2020-11-04:
Hi! @zp-yang and I are trying to get PX4 to run on Crazyflie 2.1 as well. We are new to driver development.
This is the branch we are working on: https://github.com/zp-yang/Firmware/tree/pr-crazyflie_bmi088_i2c 

Currently, the failing point we found is at  [`if (ACC_CHIP_ID != ID)` in BMI088_Accelerometer::probe()](https://github.com/zp-yang/Firmware/blob/c1936dab1fb511a34b5bb0c6b51abd537c96cc84/src/drivers/imu/bosch/bmi088/BMI088_Accelerometer.cpp#L102). 

```
int BMI088_Accelerometer::probe()
	RegisterRead(Register::ACC_CHIP_ID);

	const uint8_t ACC_CHIP_ID = RegisterRead(Register::ACC_CHIP_ID);

	if (ACC_CHIP_ID != ID) {
		DEVICE_DEBUG("unexpected ACC_CHIP_ID 0x%02x", ACC_CHIP_ID);
		return PX4_ERROR;
	}

	return PX4_OK;
```

The `ID` is [preset to 0x1E](https://github.com/zp-yang/Firmware/blob/c1936dab1fb511a34b5bb0c6b51abd537c96cc84/src/drivers/imu/bosch/bmi088/Bosch_BMI088_Accelerometer_Registers.hpp#L52), which is the reset value of ACC_CHIP_ID register, as shown on page 25 of the [datasheet](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bmi088-ds001.pdf). While the `ACC_CHIP_ID` obtained by `RegisterRead(Register::ACC_CHIP_ID)` is 0x00, hence failing condition check.

I traced back the function call flow:
```
BMI088_Accelerometer::probe()[src/drivers/imu/bosch/bmi088/BMI088_Accelerometer.cpp]
->RegisterRead(Register::ACC_CHIP_ID) [src/drivers/imu/bosch/bmi088/BMI088_Accelerometer.cpp]
->SPI::transfer(cmd, cmd, sizeof(cmd))[src/lib/drivers/device/nuttx/SPI.cpp]
->SPI::_transferhword(uint16_t *send, uint16_t *recv, unsigned len)[src/lib/drivers/device/nuttx/SPI.cpp]
->SPI_EXCHANGE(_dev, send, recv, len) [src/lib/drivers/device/nuttx/SPI.cpp]
->#define SPI_EXCHANGE(d,t,r,l) ((d)->ops->exchange(d,t,r,l))[platforms/nuttx/NuttX/nuttx/include/nuttx/spi/spi.h]
```

I don't see any logical flaw along the flow. My guess is that the issue of RegisterRead(Register::ACC_CHIP_ID) not returning 0x1E might be associated with hardware specifications in the code, e.g., pin number or bus number (I'm not an expert in this field, this is just my reasoning)

This is what I've found so far. We will keep working on it and posting updates here. If anyone has experience in driver development and is interested in working out this issue as well, please feel free to join us. It would be great if we can have someone experienced.

Something to add on:
> Waiting on I2C support for the BMI088.
@dagar Daniel said waiting on I2C support for the BMI088. Based on what I found in the [crazyflie-firmware firmware](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/drivers/bosch/src/bmi088_accel.c), BMI088 on Crazyflie supports both SPI and I2C. I was wondering if developing I2C support a must? Or as long as we fix the SPI support of BMI088 in PX4, we would be able to get Crazyflie 2.1 to fly?

TheLegendaryJedi, 2020-11-08:
Hi @JizhouChen I'm trying to flash PX4 Firmware into crazyflie2.1 too. But I really can't debug it? how are you debugging it?

Thanks

TheLegendaryJedi, 2020-11-08:
Hi @JizhouChen,

I'm trying to make it work too but I'm having a hard time figuring out on how to debug it? How could you check that the ACC_CHIP_ID is returning 0xFF?

Thanks

J22C, 2020-11-08:
> Hi @JizhouChen,
> 
> I'm trying to make it work too but I'm having a hard time figuring out on how to debug it? How could you check that the ACC_CHIP_ID is returning 0xFF?
> 
> Thanks

You need to have a debugger (the one I'm using is ST-LINK/V2) and debug the firmware with gdb

JizhouChen, 2020-11-08:
> Hi @JizhouChen,
> 
> I'm trying to make it work too but I'm having a hard time figuring out on how to debug it? How could you check that the ACC_CHIP_ID is returning 0xFF?
> 
> Thanks

You need to have a debugger (the one I'm using is ST-LINK/V2) and debug the firmware with gdb

TheLegendaryJedi, 2020-11-08:
Thanks @JizhouChen :)

I'll need to buy one then. 

Thanks

JizhouChen, 2020-11-18:
**Update 11/18:**
We have implemented the BMI088 I2C interface and got PX4 run on Craziflie (with one issue).
https://github.com/zp-yang/Firmware/tree/pr-crazyflie_bmi088_i2c

[image]

Based on our debugging result, the bmi088_i2c_main() goes through without any problem. We suspect that it has something to do with our I/O implementation and we will keep debugging it.

Please feel free to leave a comment if you have any thoughts. Thank you!

TheLegendaryJedi, 2020-11-19:
Hi @JizhouChen I've been working on an I2C version for the BMI088. I can start accel and gyro on the I2C bus but I'm having some problems with the FIFO data samples. If you want to check it. I checked yours and it's very similar we both used Daniel's SPI implementation as a base. But for some reason the fifo read is not working as expected. I tried to implement the recommended steps in the datasheet for self-test and sometimes it works and sometimes it doesn't. I hit a bit of a wall now. There was some iterations that I was able to publish correct data frames to PX4 but I can't complete accel calibration (it seems that some of the data don't have correct values when published).

https://github.com/TheLegendaryJedi/Firmware/tree/crazyflieV21

Thanks

JizhouChen, 2020-11-19:
@TheLegendaryJedi Awesome! We will check it out and update you once we find anything.

TheLegendaryJedi, 2020-11-21:
The problem I'm having now is during the fifo transfer. That is, I check the number of samples available multiply by 7 specified by the datasheet (6 for x,y,z and one for header) and the transfer always fails, it returns -1.

JizhouChen, 2020-11-23:
Hi @TheLegendaryJedi , are you on the[ PX4 Slack channel](px4.slack.com) by any chance?

TheLegendaryJedi, 2020-11-24:
> Hi @TheLegendaryJedi , are you on the PX4 Slack channel (px4.slack.com) by any chance?

Yes, I'm @GONCALO there.

I just pushed some more commits to the branch. I was able to make it work (but without the FIFO). Now I'm having problems with the attitude estimator.

If you test with my branch you should be able to calibrate accel and gyro with no problem at all.

jancoow, 2020-12-04:
Hi. 

I'm working on getting the BMI088 with I2C and FIFO to work as well for the PX4 with a crazyflight2.1 
Anyone having some progress on this implementation? It seems that reading from the FIFO buffer register is not working properly.

I've tried lowering the output data rate to 400hz because this is recommended with an I2C speed of 100kHz, but with no luck.

TheLegendaryJedi, 2020-12-04:
> Hi.
>
> I'm working on getting the BMI088 with I2C and FIFO to work as well for the PX4 with a crazyflight2.1
> Anyone having some progress on this implementation? It seems that reading from the FIFO [...]

Hi @jancoow, I've been working on it too. Are you on slack? We have a thread there regarding this.

Basically I started with fifo and then I checked that the data was all wrong. I tried with 100kHz and 400kHz and no luck.

Next I tried to read the data directly from the data registers and I was able to calibrate it. But during the gyro calibration the QGC is always saying that the drone is moving. That's because the gyro data varies a lot. The accel is all good for calibration.

But now I'm trying again to implement it with fifo because it's the correct way to do it.

I would like to know if crazyflie2.1 original firmware is using fifo or not. I need to ask on their forum.

dagar, 2022-02-03:
If anyone is motivated to bring these back to full support I'd be happy to try and assist (eg debug drivers), but otherwise I don't think I have time to fully review and test the system.