PX4/PX4-Autopilot #15363: Follow Me Mode Causing Crash

jeremyyano, 2020-07-17:
We had a crash the other day while testing the Follow Me on an X-Class quadcopter run on a CUAV V5+ on the latest stable PX4.

I've attached the log of the flight for analysis: [16_38_51.zip](https://github.com/PX4/Firmware/files/4938814/16_38_51.zip)

We had the follow target set as a Samsung Galaxy J8 running QGroundControl, and took the quad off in Stabilized mode. We hovered up to a safe position and then flipped into Follow Me mode. The quad immediately climbed very rapidly up to the 8m target, got a new position command from the Follow controller, gave a very hard yaw/pitch command, at which point the quad lost all stability and fell out of the sky.

This quadcopter has been vetted before, completing multiple 10-20min fully autonomous missions with the built in PX4 mission mode (Log example: [log_2_2020-7-9-16-23-10.zip](https://github.com/PX4/Firmware/files/4938900/log_2_2020-7-9-16-23-10.zip)). Looking at logs from those flights, even when the position setpoint changes, no truly instantaneous commands are given to the rate, attitude, or position controllers. Looking at the log for the Follow Me mode flight, when the new position setpoint command is given, instantaneous rate, attitude, and velocity commands are given to the flight controller, which we believe caused the crash due to the instantaneous massive error likely causing a D-term freakout. 

It appears that none of the max-acceleration, max-jerk, etc settings affect the Follow Me mode setpoints? Just one theory, eager to hear some others.

Thanks for your help!

dagar, 2020-07-19:
@MaEtUgR @bresch can you comment on the state of follow me?

dagar, 2020-07-19:
We should think about getting something in the simulation tests (gazebo + MAVSDK) for follow me. FYI @Jaeyoung-Lim @julianoes

MaEtUgR, 2020-07-21:
@jeremyyano Sorry to hear that. I hope your X-Class is ok. I uploaded your crash flight log to the PX4 log platform logs.px4.io for easier analysis: https://logs.px4.io/plot_app?log=3c6ba605-e3b9-4182-9a67-63b683790738

It looks like as soon as you switched to the follow target mode the heading immediately had the goal to go to zero angle relative to north, commands were given to achieve that smoothly. Once that zero heading was reached the position setpoint was commanded a several meter jump and drove all loops into their limits from which they were not able to recover quick enough anymore.

I'm currently guessing it's a combination of: follow target mode not giving feasible setpoints, vehicle tuning not being able to recover from unfeasible step jumps, motor limit reactions not being conservative enough to save the divergence. We have to look into how the horizontal setpoints for the follow target mode are generated to find out why it was so abrupt.

julianoes, 2020-07-21:
> We should think about getting something in the simulation tests (gazebo + MAVSDK) for follow me.

Who ever uses and cares about follow me should do that, agreed.

dagar, 2020-07-21:
> > We should think about getting something in the simulation tests (gazebo + MAVSDK) for follow me.
> 
> Who ever uses and cares about follow me should do that, agreed.

Yes, although certain highly visible features like this that you can slip into trivially from the QGC bar either need basic coverage for safety or need to go.

closed 2022-09-23 by junwoo091400