PX4/PX4-Autopilot #15378: no CRC lib

BazookaJoe1900, 2020-07-19:
**Describe problem solved by the proposed feature**
There are several drivers/modules that implement crc code. the code is redundant, not on a lib and not cross compiled.
I have noticed that on pr 'Battery Smbus enhancements #15369' where I used a nuttx crc code that failed to compiled on linux.

**Describe your preferred solution**
one lib to rule them all.
that lib can be as efficient as possible, cross compiled, use Hardware crc calculation, if possible. and implement needed polynomials.

**Additional context**
I have noticed the next crc8 codes:
SDP3X, DShotTelemetry, BST, crc8_dvb_s2  - "stupid" xor crc
TERARANGER, tap_esc, px4io  - has crc_table
from first look, some use same polynomial.

dagar, 2020-07-20:
We should check the numbers, but this sounds like a good idea. At a glance I think the actual overlap with NuttX proper is minimal.

BazookaJoe1900, 2020-07-23:
I tried to find all the crc8 functions on the code, here they are:

module/driver | poly | functions | type of calculation
-- | -- | -- | --
driver/SDP3X | 0x31 (x^8 + x^5 + x^4 + 1) | SDP3X::crc(const uint8_t data[], unsigned size, uint8_t   checksum) | dual loop
  |   |   |  
DShotTelemetry | seems like 07 | DShotTelemetry::crc8(const uint8_t *buf, uint8_t len) | dual loop
BST | 0xD5 | uint8_t BST::crc8(uint8_t *data, size_t len) | dual loop
lib/SMBUS | ccitt | uint8_t SMBus::get_pec(uint8_t *buff, const uint8_t len) | dual loop
lib/rc/CRSF | 0xD5 | static uint8_t crc8_dvb_s2(uint8_t crc, uint8_t a) | dual loop
lib/rc/st24 | seems like 07 | uint8_t st24_common_crc8(uint8_t *ptr, uint8_t len) | dual loop
bmp388 | ?? | static int8_t cal_crc(uint8_t seed, uint8_t data) | dual loop
px4io | ccitt (0x00, 0x07, 0x0E …) | crc_packet(struct IOPacket *pkt) | table
TERARANGER | ccitt (0x00, 0x07, 0x0E …) | internal crc8(uint8_t *p, uint8_t len) | table
nuttx/mtd | 0x00, 0x3E 0x7C… | nuttx/crc8     nuttx/crc8part | table
TAP esc | 0x00, 0xe7, 0x29 …. | crc8_esc | table