PX4/PX4-Autopilot #15395: Replacement for Discontinued LSM303D Compass

vincentpoont2, 2020-07-22:
LSM303D Compass that is being used on FMUv2 and FMUv3 has discontinued and no longer being made.  
Is there a replacement for this?

vincentpoont2, 2020-07-22:
@bkueng @dagar

BazookaJoe1900, 2020-07-22:
How about the LSM303AGR?
https://www.st.com/en/mems-and-sensors/lsm303agr.html

vincentpoont2, 2020-07-22:
> How about the LSM303AGR?
> https://www.st.com/en/mems-and-sensors/lsm303agr.html

Is there px4 driver for this?

dagar, 2020-07-22:
I would look at the lis2mdl, lsm303agr, or even RM3100 (definitely not a drop in replacement).

Our current magnetometer drivers can be found here. https://github.com/PX4/Firmware/tree/master/src/drivers/magnetometer

vincentpoont2, 2020-07-25:
@dagar Should manufacturer (Holybro) just select any of these models and ask you to add the sensor to the FMUv3 and FMUv2 [init file](https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board_sensors)?

I was thinking px4 would select one particular sensor model for all these boards. I don't quite understand this process.

vincentpoont2, 2020-07-31:
> @dagar Should manufacturer (Holybro) just select any of these models and ask you to add the sensor to the FMUv3 and FMUv2 [init [...]

@BazookaJoe1900 @davids5 Any idea?

davids5, 2020-07-31:
> I was thinking px4 would select one particular sensor model for all these boards.

@dagar I think @vincentpoont2 has an excellent point. If we want to avoid more issues. We should pick a part we are happy with performance wise and suggest the chip orientation WRT direction of flight and Z.

vincentpoont2, 2020-09-20:
> > I was thinking px4 would select one particular sensor model for all these boards.
> 
> @dagar I think @vincentpoont2 has an excellent point. If we want to avoid more issues. We should pick a part we are happy with performance wise and suggest the chip orientation WRT direction of flight and Z.

Hi all, I would like to follow up on this. 

There are people still interested in buying the FMUv2 & FMUv3 products due to the lower cost and other reasons.
If Holybro change the compass to lis2mdl or lsm303agr, since the signals pin are not pin to pin compatible to the original LSM303D, is there any reference on how this can be achieved and still be software compatible ?

davids5, 2020-09-21:
Without a way to know for sure the board is not an old one (there are [hacks to detect the some difference](https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/src/init.c#L168) but it is sketchy at best  ),  We could extend this hack (Pull  some pins to opposite states the ALL other boards), or give it a new bootloader ID,  - OR should we just try start the new device and call it day?

@vincentpoont2 - Both the devices you are choosing have distinct WHOAMI so we have a chance of just starting them. If the mag is defective it will not get the board type correct. 

@dagar Opinion?

vincentpoont2, 2022-09-16:
Holybro FMUv2 and FMUv3 hardware has been discontinued.

closed 2022-09-16 by vincentpoont2