PX4/PX4-Autopilot #15400: make px4_fmu-v2_default Fail

Pigwomaniak, 2020-07-22:
**After `make px4_fmu-v2_default` command this fail comes,**

`[219/221] Linking CXX executable px4_io-v2_default.elf
FAILED: px4_io-v2_default.elf 
: && /home/maciek/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++  -mcpu=cortex-m3 -mthumb -Os -DNDEBUG  --specs=nosys.specs -fuse-ld=gold platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj  -o px4_io-v2_default.elf -L/home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware/external/Install/lib -nostartfiles  -nodefaultlibs  -nostdlib  -nostdinc++  -fno-exceptions  -fno-rtti  -Wl,--script=/home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware/NuttX/nuttx-config/scripts/script.ld  -Wl,-Map=px4_io-v2_default.map  -Wl,--warn-common  -Wl,--gc-sections  -Wl,--print-memory-usage  -Wl,--start-group  NuttX/apps/libapps.a  NuttX/nuttx/arch/arm/src/libarch.a  NuttX/nuttx/binfmt/libbinfmt.a  NuttX/nuttx/libs/libc/libc.a  NuttX/nuttx/boards/libboards.a  NuttX/nuttx/libs/libxx/libxx.a  NuttX/nuttx/drivers/libdrivers.a  NuttX/nuttx/fs/libfs.a  NuttX/nuttx/mm/libmm.a  NuttX/nuttx/sched/libsched.a  -Wl,--end-group  -lm  -lgcc  src/modules/px4iofirmware/libpx4iofirmware.a  NuttX/apps/libapps.a  platforms/nuttx/src/px4/stm/stm32f1/io_pins/libarch_io_pins.a  src/lib/mixer/libmixer.a  src/lib/mixer/HelicopterMixer/libHelicopterMixer.a  src/lib/mixer/MultirotorMixer/libMultirotorMixer.a  src/lib/mixer/NullMixer/libNullMixer.a  src/lib/mixer/SimpleMixer/libSimpleMixer.a  src/lib/mixer/Mixer/libMixer.a  src/lib/rc/librc.a  src/lib/output_limit/liboutput_limit.a  NuttX/nuttx/arch/arm/src/libarch.a  boards/px4/io-v2/src/libdrivers_board.a  platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a  src/lib/parameters/libparameters.a  src/lib/parameters/flashparams/libflashparams.a  NuttX/nuttx/arch/arm/src/libarch.a  boards/px4/io-v2/src/libdrivers_board.a  platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a  src/lib/parameters/libparameters.a  src/lib/parameters/flashparams/libflashparams.a  NuttX/nuttx/libs/libxx/libxx.a  NuttX/nuttx/libs/libc/libc.a  NuttX/nuttx/drivers/libdrivers.a  NuttX/nuttx/fs/libfs.a  NuttX/nuttx/mm/libmm.a  NuttX/nuttx/sched/libsched.a  -lm  -lgcc  src/lib/parameters/tinybson/libtinybson.a  platforms/common/libpx4_platform.a  msg/libuorb_msgs.a && :
/usr/bin/arm-none-eabi-ld.gold: --print-memory-usage: unknown option
/usr/bin/arm-none-eabi-ld.gold: use the --help option for usage information
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[470/735] Building CXX object src/modu....dir/checks/manualControlCheck.cpp.obj
FAILED: external/Stamp/px4io_firmware/px4io_firmware-build 
cd /home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware && /usr/bin/cmake --build .
[480/735] Building CXX object src/modu...dules__commander.dir/Commander.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:219: px4_fmu-v2_default] Error 1`

It happeds also for `make px4_fmu-v3_default` but the simulation `make px4_sitl jmavsim` works fine
I would be grateful for any suggestions

dagar, 2020-07-22:
It looks like the problem is you're linking with `arm-none-eabi-ld.gold` which apparently doesn't support the `--print-memory-usage` output NuttX uses.

Short term you can comment it out here.  https://github.com/PX4/Firmware/blob/e4f65f81f72ab6c2bb599bece9e55317826d981c/platforms/nuttx/CMakeLists.txt#L127

dagar, 2020-07-22:
https://github.com/PX4/Firmware/pull/15401

Pigwomaniak, 2020-07-22:
The output error now looks like that:
`[219/221] Linking CXX executable px4_io-v2_default.elf
FAILED: px4_io-v2_default.elf 
: && /home/maciek/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++  -mcpu=cortex-m3 -mthumb -Os -DNDEBUG  --specs=nosys.specs -fuse-ld=gold platforms/nuttx/CMakeFiles/px4.dir/__/common/empty.c.obj  -o px4_io-v2_default.elf -L/home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware/external/Install/lib -nostartfiles  -nodefaultlibs  -nostdlib  -nostdinc++  -fno-exceptions  -fno-rtti  -Wl,--script=/home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware/NuttX/nuttx-config/scripts/script.ld  -Wl,-Map=px4_io-v2_default.map  -Wl,--warn-common  -Wl,--gc-sections  -Wl,--start-group  NuttX/apps/libapps.a  NuttX/nuttx/arch/arm/src/libarch.a  NuttX/nuttx/binfmt/libbinfmt.a  NuttX/nuttx/libs/libc/libc.a  NuttX/nuttx/boards/libboards.a  NuttX/nuttx/libs/libxx/libxx.a  NuttX/nuttx/drivers/libdrivers.a  NuttX/nuttx/fs/libfs.a  NuttX/nuttx/mm/libmm.a  NuttX/nuttx/sched/libsched.a  -Wl,--end-group  -lm  -lgcc  src/modules/px4iofirmware/libpx4iofirmware.a  NuttX/apps/libapps.a  platforms/nuttx/src/px4/stm/stm32f1/io_pins/libarch_io_pins.a  src/lib/mixer/libmixer.a  src/lib/mixer/HelicopterMixer/libHelicopterMixer.a  src/lib/mixer/MultirotorMixer/libMultirotorMixer.a  src/lib/mixer/NullMixer/libNullMixer.a  src/lib/mixer/SimpleMixer/libSimpleMixer.a  src/lib/mixer/Mixer/libMixer.a  src/lib/rc/librc.a  src/lib/output_limit/liboutput_limit.a  NuttX/nuttx/arch/arm/src/libarch.a  boards/px4/io-v2/src/libdrivers_board.a  platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a  src/lib/parameters/libparameters.a  src/lib/parameters/flashparams/libflashparams.a  NuttX/nuttx/arch/arm/src/libarch.a  boards/px4/io-v2/src/libdrivers_board.a  platforms/nuttx/src/px4/stm/stm32f1/hrt/libarch_hrt.a  src/lib/parameters/libparameters.a  src/lib/parameters/flashparams/libflashparams.a  NuttX/nuttx/libs/libxx/libxx.a  NuttX/nuttx/libs/libc/libc.a  NuttX/nuttx/drivers/libdrivers.a  NuttX/nuttx/fs/libfs.a  NuttX/nuttx/mm/libmm.a  NuttX/nuttx/sched/libsched.a  -lm  -lgcc  src/lib/parameters/tinybson/libtinybson.a  platforms/common/libpx4_platform.a  msg/libuorb_msgs.a && :
/usr/bin/arm-none-eabi-ld.gold: warning: creating a segment to contain the file and program headers outside of any MEMORY region
/usr/bin/arm-none-eabi-ld.gold: error: section .text overflows end of region flash
/usr/bin/arm-none-eabi-ld.gold: error: section .init_section overflows end of region flash
/usr/bin/arm-none-eabi-ld.gold: error: section .ARM.extab overflows end of region flash
/usr/bin/arm-none-eabi-ld.gold: error: section .ARM.exidx overflows end of region flash
/usr/bin/arm-none-eabi-ld.gold: error: section .data overflows end of region flash
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[714/735] Linking CXX static library src/modules/sensors/libmodules__sensors.a
FAILED: external/Stamp/px4io_firmware/px4io_firmware-build 
cd /home/maciek/Firmware/build/px4_fmu-v2_default/external/Build/px4io_firmware && /usr/bin/cmake --build .
[717/735] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/modules__mavlink_unity.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:219: px4_fmu-v2_default] Error 1
`

dagar, 2020-07-22:
Thanks, it looks like we shouldn't be using the gold linker at all for NuttX until someone works through these issues. I'll update the PR.

commit 6bd0307e (2020-07-22): cmake: only allow gold linker for posix builds

 - the gold linker doesn't currently work for NuttX builds
 - NuttX skip --print-memory-usage if using the GOLD linker
 - fixes #15400
PR #15401 (2020-07-23): cmake: NuttX skip --print-memory-usage if using the GOLD linker
 - fixes #15400

@Pigwomaniak can you give this a try?
commit 9426c68a (2020-07-23): cmake: only allow gold linker for posix builds

 - the gold linker doesn't currently work for NuttX builds
 - NuttX skip --print-memory-usage if using the GOLD linker
 - fixes #15400
closed 2020-07-23 by dagar