PX4/PX4-Autopilot #15408: Takeoff ramp broken

ThomasRigi, 2020-07-23:
**Describe the bug**
With MPC_TKO_RAMP_T = 5s the drone does not take off anymore when starting a mission. With MPC_TKO_RAMP_T = 3s it still takes off, but only after 4s not doing anything and then taking off with a ramp time of 1s.

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout current master (aa7735c1c0)
2. Launch SITL. I used `make px4_sitl gazebo_standard_vtol`
3. Set MPC_TKO_RAMP_T = 5 (1st log)
4. Load a mission. E.g. [square vtol.zip](https://github.com/PX4/Firmware/files/4967582/square.vtol.zip)
5. Start the mission
6. The drone does not take off.
7. Repeat with MPC_TKO_RAMP_T = 3 (2nd log) and MPC_TKO_RAMP_T = 1 (3rd log) and observe the differences

**Expected behavior**
The drone should take off gently respecting MPC_TKO_RAMP_T.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=0aa9b360-1230-4944-9159-4fb03aabd9d3
https://logs.px4.io/plot_app?log=44529271-2e29-4827-a09a-64d78b725c6d
https://logs.px4.io/plot_app?log=f698d92c-6636-496b-921a-5bc2c3d8a02f

**Drone (please complete the following information):**
Tested with gazebo standard_vtol

MaEtUgR, 2020-07-27:
I'm convinced the problem is this part: https://github.com/PX4/Firmware/blob/7354e3989335dfb38cc41216182f48fe84d165c7/src/lib/flight_tasks/tasks/AutoLineSmoothVel/FlightTaskAutoLineSmoothVel.cpp#L335-L342

@ThomasRigi Could you try the same experiment with that block commented and see if that solves your issues?

ThomasRigi, 2020-08-03:
@MaEtUgR With these lines commented, the drone takes off with a nice ramp, but obviously ignores MPC_TKO_SPEED.

https://logs.px4.io/plot_app?log=c376d64e-d23d-4052-8149-66a6bf6b8ac8

In our use case we don't really care about MPC_TKO_SPEED as we prefer to have one long smooth ramp until the final speed. So commenting these lines works for us, but it's not a nice fix for people who rely on MPC_TKO_SPEED.

closed 2020-08-10 by bresch