PX4/PX4-Autopilot #15409: [VTOL] Not yawing towards VTOL Takeoff & Transition waypoint

ThomasRigi, 2020-07-23:
**Describe the bug**
The VTOL is no longer yawing towards the VTOL Takeoff & Transition waypoint. This causes the mission to fail, triggering RTL (which then itself has a bug).
@RomanBapst 

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout current master (aa7735c)
2. `make px4_sitl gazebo_standard_vtol`
3. Load a mission with a VTOL Takeoff & Transition waypoint, e.g. [square vtol.zip](https://github.com/PX4/Firmware/files/4967632/square.vtol.zip)
4. Start the mission and observe.

**Expected behavior**
The drone should yaw towards the waypoint and then transition.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=44529271-2e29-4827-a09a-64d78b725c6d (RTL)

**Drone (please complete the following information):**
Gazebo standard_vtol

dagar, 2020-07-23:
Do you happen to know when this last worked as expected/desired?

ThomasRigi, 2020-07-23:
Most recent I have tested is on PR #15034 (June 04). Something in between must have gone wrong.

sfuhrer, 2020-08-03:
It was this one: https://github.com/PX4/Firmware/pull/15358.

closed 2020-08-04 by bresch