PX4/PX4-Autopilot #15424: Compile Error : px4_sitl_rtps using ros2

stmoon, 2020-07-27:
Hi,

When I compile px4_sitl_rtps with ros2 (dashing),  I found the compile error like below

```
[  0%] Building CXX object CMakeFiles/micrortps_agent.dir/RtpsTopics.cpp.o
In file included from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/microRTPS_timesync.h:46:0,
                 from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:38,
                 from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:34:
/root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/timesync_Publisher.h:53:24: error: ‘px4’ does not name a type
 using timesync_msg_t = px4::msg::timesync;
```

Without ros environment, it works. However, with ros environment, I think that the package name is wrong.

How can I solve this problem?

TSC21, 2020-07-27:
What are the steps to reproduce this?

stmoon, 2020-07-27:
I just follow the instruction  (https://dev.px4.io/master/en/middleware/micrortps.html)

```
$ mkdir -p ~/px4_ros_com_ros2/src
$ git clone https://github.com/PX4/px4_ros_com.git ~/px4_ros_com_ros2/src/px4_ros_com # clones the master branch
$ git clone https://github.com/PX4/px4_msgs.git ~/px4_ros_com_ros2/src/px4_msgs
```bash

and run 
```
$ colcon build
$ . install/setup.bash
```

Then compile firmware

```
$ make px4_fmu-v3_rtps
```

and change the directory to compile agent.

```
$ cmake ..
$ build
```

I use dashing. and the branch of firmware is master.

```
1384/1384] Creating /root/test/Firm.../px4_fmu-v3_rtps/px4_fmu-v3_rtps.px4
root@stmoon-XPS-13-9370:~/test/Firmware# goagent 
root@stmoon-XPS-13-9370:~/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build# cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build
root@stmoon-XPS-13-9370:~/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/build# make 
[  0%] Building CXX object CMakeFiles/micrortps_agent.dir/RtpsTopics.cpp.o
In file included from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/microRTPS_timesync.h:46:0,
                 from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.h:38,
                 from /root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/RtpsTopics.cpp:34:
/root/test/Firmware/build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent/timesync_Publisher.h:53:24: error: ‘px4’ does not name a type
 using timesync_msg_t = px4::msg::timesync;

```

stmoon, 2020-07-27:
@TSC21  Is the agent code in build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent after firmware not used for ros2?

TSC21, 2020-07-27:
> @TSC21 Is the agent code in build/px4_sitl_rtps/src/modules/micrortps_bridge/micrortps_client/micrortps_agent after firmware not used for ros2?

No. The templates are adjusted according to the data types. The agent inside px4_ros_com is the one that is built inside a ROS2 colcon workspace.

closed 2020-07-28 by stmoon