PX4/PX4-Autopilot #15435: pwm command does not work for UGV (50002)

stmoon, 2020-07-28:
Hi, 

I bought Traxxas Stampede VXL to apply to PX4 (px4-fmu-v3-default for pixhawk 2.1)
and I changed air frame  (50002, traxxas_stampede_2wd)

Before the operation test, I just try to test pwm using "pwm test -c 24 -p 1700"
However, when pwm command is entered, there is no reaction.

It only works once after booting by PWM_MIN.  

Why is the pwm command not working?

stmoon, 2020-08-01:
Does anyone have any idea?

Jaeyoung-Lim, 2021-07-13:
@stmoon There has been quite some fixes recently, and rover should be in a working state now.

stmoon, 2021-07-14:
Thanks! I will try it again !.

chfriedrich98, 2024-09-13:
Cleanup rover related issues:
The rover support in PX4 has been completely revamped since this issue has been opened.

closed 2024-09-13 by chfriedrich98