PX4/PX4-Autopilot #15446: uORB messages are not getting updated - PX4  SITL (v1.10.2) with jMAVSim

arun-math-code, 2020-07-29:
**Describe the bug**
After launching the PX4 SITL with jMAVSim, after few minutes, the uORB messages (say sensor_accel) stopped updating. 
I am using listener command in the PX4 SITL terminal to listen to uORB messages. After 2-3 minutes, I could observe that the uORB messages stopped updating. The uORB timestamp was still showing the older timestamp value and that's how I concluded uORB messages are not getting updated. Surprisingly it was still showing updated 0 seconds ago even though I tried it several minutes later!  Is that indicate the SITL time is somehow frozen? Attaching the screenshots below.

[image]
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Clone and then checkout v1.10.2
     - git clone https://github.com/PX4/Firmware.git
     - cd Firmware/
     - git checkout v1.10.2
2. Make sure QGroundCrontrol is closed.
3. Build and launch SITL with jMAVSim ( make px4_sitl jmavsim )
4. Keep the terminal idle for few minutes after jMAVSim launched
5. listen to 'sensor_accel' ( listener sensor_accel ) 

**Expected behavior**
'sensor_accel' should get updated. Even if it was showing old timestamp data, it should not show updated 0 seconds ago.

**Additional context**
When you observe this behavior, if you open QGroundControl you can see that it won't connect to the SITL.

Ankur1014, 2020-07-29:
I am having the same observations on 1.10.2. It appears that the whole PX4 SITL is frozen

arun-math-code, 2020-08-02:
I could see the issue in master as well !

julianoes, 2020-11-24:
@arun-math-code could you check again if this has been fixed in the meantime?