PX4/PX4-Autopilot #15448: make px4_sitl_default sitl_gazebo fails to build

Reuben-Isaac, 2020-07-30:
I used to be able to run this command but recently I installed Unreal Engine to test AirSim and now this Gazebo build command fails with the following error. Could anyone tell me what is happening? 

I am running PX4 version 1.10.1
```
undergrad@asg-a32249:~/Firmware$ DONT_RUN=1 make px4_sitl_default gazebo
[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.58") found components: system thread filesystem chrono date_time atomic 
CMake Warning (dev) at /usr/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:30 (include)
  CMakeLists.txt:32 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.40.0") found components: thread signals system filesystem program_options regex iostreams date_time chrono atomic 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Found Boost: /usr/include (found version "1.65.1")  
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Found UUID: TRUE  
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Found UUID: TRUE  
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
CMake Error at CMakeLists.txt:34 (find_program):
  Could not find px4 using the following names:

-- Configuring incomplete, errors occurred!
See also "/home/undergrad/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/undergrad/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/undergrad/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/undergrad/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/undergrad/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
```

haohuachang, 2020-08-03:
@Reuben-Isaac I also encountered a similar problem, I tried to do the following：

1. update sitl_gazebo code  to header commit on this [branch](https://github.com/PX4/sitl_gazebo),you only update or rebase your px4 master

1. update qt
   
```bash
brew unlink qt@4
brew restall qt
```

julianoes, 2020-08-03:
This fixes it for master https://github.com/PX4/sitl_gazebo/pull/550.

You would have to apply these changes in Tools/sitl_gazebo for 1.10.1.

bamboosdu, 2020-09-27:
> This fixes it for master [PX4/sitl_gazebo#550](https://github.com/PX4/sitl_gazebo/pull/550).
> 
> You would have to apply these changes in Tools/sitl_gazebo for 1.10.1.

maybe a stupid question. How to apply these changes ? Thanks!

julianoes, 2020-09-28:
@bamboosdu by now these changes should be in master PX4. So if you try the latest PX4 I expect it to be solved.

closed 2021-01-17 by Jaeyoung-Lim