PX4/PX4-Autopilot #15450: Sometimes takeoff fails in CI tests

julianoes, 2020-07-30:
When working on the MAVSDK SITL CI tests I often run into the case where a takeoff fails completely, or the landed state toggles back and forth between finally taking off.

Here is an example where takeoff doesn't happen at all:
https://logs.px4.io/plot_app?log=0f20b944-0704-4cbf-b499-1aec5bf6f9d6

And here are examples where takeoff on the second try seems to work:
https://logs.px4.io/plot_app?log=972ffaf8-da85-4142-a469-7195e0ba342f
https://logs.px4.io/plot_app?log=f13ba39e-cab0-4feb-8f6a-f7b8a506bc9c
https://logs.px4.io/plot_app?log=bab77966-e501-4923-9181-e922063a4109

You can see how there is:
```
[commander] Takeoff detected
[commander] Landing detected
[commander] Takeoff detected
```

FYI @MaEtUgR @dagar

MaEtUgR, 2020-07-30:
https://github.com/PX4/Firmware/issues/15408

MaEtUgR, 2020-11-09:
This was solved by #15509

closed 2020-11-09 by MaEtUgR