PX4/PX4-Autopilot #15455: Failsafe time vs external vision position loss

PLMR-FR, 2020-07-30:
**Describe the bug**
The failsafe time when external vision position stops to stream is too long (5s).

**To Reproduce**
Setup : 
- Based on FmuV4 hardware
- 2 companion computers (on Telem 1 and 2, external vision is on Telem 1)
- Software V1.11.0-rc1
- No RC

**Expected behavior**
I was testing how the drone lands when it loses the position from external vision, and after switching off the external vision, we have to wait 5 seconds before the autopilot performes a blind land.
I can't find any parameter to reduce this time (ideal would be 0.5sec)
When you fly indoor close to objects and you lose eternal vision, 5 seconds are way too long and the position diverges too much.

**Log Files and Screenshots**
Here is a log of the issue : https://review.px4.io/plot_app?log=9aa711ef-f168-456e-af83-0e42afafd495

How can I change this delay of 5 seconds ? I presume it's in mc_pos_control module but now I haven't found it.

Thanks

PLMR

julianoes, 2020-08-03:
Can you check if you find the timeout in commander? Or ekf2?

PLMR-FR, 2020-08-04:
Hi,

It seems that the timeout (reset_timeout_max) is in ekf2 and set at the value of 7 seconds. (src/lib/ecl/EKF/common.h , line 334)

`unsigned reset_timeout_max{7000000};	///< maximum time we allow horizontal inertial dead reckoning before attempting to reset the states to the measurement or change _control_status if the data is unavailable (uSec)`

And it is used in src/lib/ecl/EKF/control.cpp line 330
```
else if ((_control_status.flags.ev_pos || _control_status.flags.ev_vel)
		   && isTimedOut(_time_last_ext_vision, (uint64_t)_params.reset_timeout_max)) {

		// Turn off EV fusion mode if no data has been received
		stopEvFusion();
		ECL_INFO_TIMESTAMPED("vision data stopped");

	}
```

I will try smaller values.

PLMR