PX4/PX4-Autopilot #15461: feature request : calculate terrain alt (AGL alt) during flight

moreba1, 2020-07-30:
Hi

For flying in some locations like mountains and valley, it is better that we have terrain alt. method of calculating terrain alt in px4 is using sonar and lasers. but there is a different method that we can use to monitor and manually avoid collision with the ground.
we have AMSL alt from GPS and the alt of the ground can be extracted from internet free sources with qgroundcontrol and save information area of mission it and upload to sd card.
so terrain alt can be calculated during flight and display in the ground station.

bkueng, 2020-07-31:
Hi

I'm currently working on that, namely terrain data upload to the SD card, which then can be used for example during RTL.

moreba1, 2020-08-11:
https://github.com/PX4/Firmware/issues/14265

closed 2020-08-11 by moreba1