PX4/PX4-Autopilot #15464: PX4 external pose estimation

zhanqx1024, 2020-07-31:
Hello, I want to use vision for the external position estimation, so I set the Parameters EKF2_AID_MASK in QGroundControl. I set the bits in position 3 to enable vision position fusion, and the bits in position 4 to enable vision yaw fusion.

I get the position estimation from my vision algorithm. Then I publish the position and orientation data via the publish node "/mavros/vision_pose/pose".

In the subscribe node "/mavros/imu/data", I get the orientation data from Imu. (I put a magnet next to the magnetometer, so the orientation data from node "/mavros/imu/data" is different from node "/mavros/vision_pose/pose".

In the subscribe node "/mavros/local_position/pose", I get the position data and orientation data. The position data is the same with that from the node "/mavros/vision_pose/pose". But the yaw data is almost the same with the that from node  "/mavros/imu/data". So how can I set the Parameters in QGroundControl, or modify the  PX4 source code, or do something to make the yaw data the same with the yaw data from node "/mavros/vision_pose/pose". 

My email address is 1930739@tongji.edu.cn, I am very pleasure of your suggestions. Thank you very much!

[image]

bresch, 2020-08-03:
You first need to make sure the yaw vision data is actually fused by the EKF (https://github.com/PX4/Firmware/blob/6923d9eb644c21ec337b65e75bd8024209a86a5b/msg/estimator_status.msg#L41)

zhanqx1024, 2020-08-03:
Thanks for your reply, I find the control_mode_flags in the _estimator_status_0 of the .ulg file is 12931 and 129803,this number is  decimal, I converted it to binary,they are 11001010000011 and 1100100000011,the 13st bitmask  of them are  all 1,so the CS_EV_YAW is true. this is the flight review  https://logs.px4.io/plot_app?log=c45d2611-6295-417b-9310-ef7096906d87

zhanqx1024, 2020-08-03:
Thanks for your reply, I find the control_mode_flags in the _estimator_status_0 of the .ulg file is 12931 and 12803,this number is decimal, I converted it to binary,they are 11001010000011 and 1100100000011,the 13st bitmask of them are all 1,so the CS_EV_YAW is true. this is the flight review https://logs.px4.io/plot_app?log=c45d2611-6295-417b-9310-ef7096906d87

bresch, 2020-08-07:
@kamilritz Any idea here?

kamilritz, 2020-08-07:
The internal attitude is following the vision attitude. My guess is that the yaw angle inside the imu message is not the one of the actual ekf state.
[image]

zhanqx1024, 2020-08-10:
> The internal attitude is following the vision attitude. My guess is that the yaw angle inside the imu message is not the one of the actual ekf state.
> [image]

Thanks for your reply. Sorry, the flight review which I upload before is the correct data. Here is another wrong data, https://logs.px4.io/plot_app?log=db71477d-314c-4f91-bbfd-68b3eb53bb7f

I put a magnet  next to the flight controller, meanwhile the drone is standing still on the ground. And I have chosen the vision fusion(the EKF2_AID_MASK is 24), I plot  the "vehicle_attitude" and "vehicle_visual_odometry", like this 
[image]
[image]

My Problem is that the internal attitude of yaw is following the IMU message, but not following the vision yaw, you can see from the picture of q0 and q3. I find that the q1 and q2 also have a big fluctuation which means that the fluctuations in pitch and roll are also very large. I guess that the magnets may interfere with the gyroscope and accelerometer. But our real flight scene has many steel frame structures. As shown in the figures below, it only caused magnetic shielding, but not affect the gyroscope and accelerometer.
[image]
[image]

kamilritz, 2020-08-11:
It seems that the autopilot stopped receiving any VIO data after 124 seconds. 7 seconds later it will stop fusing VIO data as designed and fallback to the magnetometer sensor. After this you start disturbing the mag. At the end of the log it jumps back to use the VIO data, once it starts to receive VIO data again.
[image]

You will need to investigate why the VIO data stops to be received in between. But that is not my field of expertise.