PX4/PX4-Autopilot #15466: MRO X2.1-777

taileron, 2020-08-01:
**Describe the bug**
after flashing PX4 to the FC, QGC doesn´t see it anymore via USB

**To Reproduce**
Steps to reproduce the behavior:
unpack it
QGC connects to Arduplane (ChibiOS)
mro_x21-777_default.px4 flash successful to:
Device: ArduPilot ChibiOS
Version: 5
Board ID:136 
(prebuild bin or from source)
after the flash procedure and reboot to PX4 
QGC doesn´t find it anymore via USB
there is also nothing special to find on the SD

**Expected behavior**
QGC should connect to it after reboot

Is there an adapted boot loader available to change it from ChibiOS to NuttX ?

dagar, 2020-08-03:
I'll take a look, it might be QGC not recognizing it for some reason.

taileron, 2020-08-03:
it seems that the boot process does not finish, the led which flashes fast for a short time before the board is ready does not stop flashing, like when the update of the IO board does not finish.
Flashed back to ChibiOS ArduPlane QGC connects immediately without problems in the moment when the led stops blinking.

taileron, 2020-08-06:
@dagar can I do tests with changed start scripts to look in which module the start procedure stops? 
e.g. flash a version without io-unit

dagar, 2020-08-09:
Should be fixed in https://github.com/PX4/Firmware/pull/15521.

closed 2020-08-09 by dagar