PX4/PX4-Autopilot #15471: Failed to transition from TAKEOFF to MISSION

nrogelio, 2020-08-02:
While testing PR #15083 the vehicle was commanded to execute a mission, took off but failed to transition from Takeoff to Mission.

A workaround was to pause the mission and then continue the mission.

Vehicle used: Holybro S500 with Pixhawk 4 and Drotek GPS.

Mission with RTL
In 4/5 missions the vehicle tookoff and didn't went to the next waypoint, I think is related to the issue of the auto-takeoff, land flights. I needed to pause the mission or switch different flight mode and then click continue mission.
https://review.px4.io/plot_app?log=d2dc82d7-0024-4562-b8dc-2edf5e4bbf22
https://review.px4.io/plot_app?log=b1762b52-0ce7-4231-8f66-004e334ccdf2
https://review.px4.io/plot_app?log=7d6db655-f099-4d24-9de9-878498a6cace <- log with no issue
https://review.px4.io/plot_app?log=dbb8da1c-8f3f-4e56-91a3-ee9ed80b6585
https://review.px4.io/plot_app?log=936188a2-8917-4c4e-9820-a519e949d8ba

divyanshu1234, 2021-07-04:
I am facing the same issue with my multicopter running PX4 release/1.12 on a Cube Orange. Is any work being done to fix this?

julianoes, 2023-04-03:
This is fairly old and I haven't seen recent reports so I'm going to close the issue.

closed 2023-04-03 by julianoes