PX4/PX4-Autopilot #15476: fails to start gazebo on Mac

ghostSAS, 2020-08-03:
Hi guys, 

I tried to start the gazebo on my Mac but failed. I have pulled the latest version of the firmware. The error is reported here:

> $ make px4_sitl gazebo_tailsitter
> [0/5] Performing build step for 'sitl_gazebo'
> [2/95] Building CXX object CMakeFiles/std_msgs.dir/Int32.pb.cc.o
> FAILED: CMakeFiles/std_msgs.dir/Int32.pb.cc.o 
> /Library/Developer/CommandLineTools/usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.89 -DLIBBULLET_VERSION_GT_282 -Dstd_msgs_EXPORTS -I/Users/ys/Zeng_Lab/Tailsitter/Firmware/Tools/sitl_gazebo/include -I. -I/usr/local/include/eigen3/eigen3 -I/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs -I/Users/ys/Zeng_Lab/Tailsitter/Firmware/mavlink/include -I/Users/ys/Zeng_Lab/Tailsitter/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/local/Cellar/libffi/3.3/include -I/usr/local/Cellar/gstreamer/1.16.2/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.64.4_1/include -I/usr/local/Cellar/glib/2.64.4_1/include/glib-2.0 -I/usr/local/Cellar/glib/2.64.4_1/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.44/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include -isystem /usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9 -isystem /usr/local/Cellar/bullet/2.89/include/bullet -isystem /usr/local/Cellar/bullet/2.89/include -isystem /usr/local/include/simbody -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat6/6.2.0_1/include/sdformat-6.2 -isystem /usr/local/include/ignition/math4 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_8/include/OGRE/Paging -isystem /usr/local/include/ignition/transport4 -isystem /usr/local/include/ignition/msgs1 -isystem /usr/local/include/ignition/common1 -isystem /usr/local/include/ignition/fuel_tools1 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.4.0/include/opencv4 -isystem /usr/local/Cellar/fcl/0.6.1/include -isystem /usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ffmpeg/4.3.1/include -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -fPIC -Wno-deprecated-declarations -Wno-address-of-packed-member -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -I/usr/local/Cellar/ffmpeg/4.3.1/include -std=gnu++17 -MD -MT CMakeFiles/std_msgs.dir/Int32.pb.cc.o -MF CMakeFiles/std_msgs.dir/Int32.pb.cc.o.d -o CMakeFiles/std_msgs.dir/Int32.pb.cc.o -c Int32.pb.cc
> In file included from Int32.pb.cc:4:
> In file included from ./Int32.pb.h:23:
> In file included from /usr/local/include/google/protobuf/io/coded_stream.h:143:
> In file included from /usr/local/include/google/protobuf/stubs/strutil.h:37:
> In file included from /usr/local/include/google/protobuf/stubs/stringpiece.h:152:
> In file included from /usr/local/include/google/protobuf/stubs/hash.h:39:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/unordered_map:407:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__hash_table:18:
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:317:9: error: no member named 'signbit' in the global namespace
> using ::signbit;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:318:9: error: no member named 'fpclassify' in the global namespace
> using ::fpclassify;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:319:9: error: no member named 'isfinite' in the global namespace; did you mean 'finite'?
> using ::isfinite;
>       ~~^
> /usr/local/include/math.h:749:12: note: 'finite' declared here
> extern int finite(double)
>            ^
> In file included from Int32.pb.cc:4:
> In file included from ./Int32.pb.h:23:
> In file included from /usr/local/include/google/protobuf/io/coded_stream.h:143:
> In file included from /usr/local/include/google/protobuf/stubs/strutil.h:37:
> In file included from /usr/local/include/google/protobuf/stubs/stringpiece.h:152:
> In file included from /usr/local/include/google/protobuf/stubs/hash.h:39:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/unordered_map:407:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__hash_table:18:
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:320:9: error: no member named 'isinf' in the global namespace
> using ::isinf;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:321:9: error: no member named 'isnan' in the global namespace
> using ::isnan;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:322:9: error: no member named 'isnormal' in the global namespace
> using ::isnormal;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:323:7: error: no member named 'isgreater' in the global namespace; did you mean '::std::greater'?
> using ::isgreater;
>       ^~
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/functional:731:29: note: '::std::greater' declared here
> struct _LIBCPP_TEMPLATE_VIS greater : binary_function<_Tp, _Tp, bool>
>                             ^
> In file included from Int32.pb.cc:4:
> In file included from ./Int32.pb.h:23:
> In file included from /usr/local/include/google/protobuf/io/coded_stream.h:143:
> In file included from /usr/local/include/google/protobuf/stubs/strutil.h:37:
> In file included from /usr/local/include/google/protobuf/stubs/stringpiece.h:152:
> In file included from /usr/local/include/google/protobuf/stubs/hash.h:39:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/unordered_map:407:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__hash_table:18:
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:324:7: error: no member named 'isgreaterequal' in the global namespace; did you mean '::std::greater_equal'?
> using ::isgreaterequal;
>       ^~
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/functional:760:29: note: '::std::greater_equal' declared here
> struct _LIBCPP_TEMPLATE_VIS greater_equal : binary_function<_Tp, _Tp, bool>
>                             ^
> In file included from Int32.pb.cc:4:
> In file included from ./Int32.pb.h:23:
> In file included from /usr/local/include/google/protobuf/io/coded_stream.h:143:
> In file included from /usr/local/include/google/protobuf/stubs/strutil.h:37:
[... 90 lines cut ...]
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:326:9: error: no member named 'islessequal' in the global namespace
> using ::islessequal;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:327:9: error: no member named 'islessgreater' in the global namespace
> using ::islessgreater;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:328:9: error: no member named 'isunordered' in the global namespace
> using ::isunordered;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
> using ::isunordered;
>       ~~^
> 13 errors generated.
> ninja: build stopped: subcommand failed.
> [1/5] Generating ../../logs
> FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
> cd /Users/ys/Zeng_Lab/Tailsitter/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.18.1/bin/cmake --build /Users/ys/Zeng_Lab/Tailsitter/Firmware/build/px4_sitl_default/build_gazebo -- -j2
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:226: px4_sitl] Error 1

Per the report, there are multiple variables that are not defined in the global namespace, which is quite weird.  
Also the jMAVSim is able to work without any issue. 

Thanks in advance for your kind help.

julianoes, 2020-08-04:
Please provide the full output after a `make clean`.

ghostSAS, 2020-08-04:
Please correct me if I understand you wrong. I first `make clean`, then `ulimit -S -n 500` to make sure the number of executed files is not an issue, then `make px4_sitl gazebo_tailsitter` again. And the output is 

> $ make px4_sitl gazebo_tailsitter
> -- PX4 version: v1.11.0-rc1-171-gf0dc76a8cf
> -- PX4 config file: /Users/ys/Zeng_Lab/Tailsitter/Firmware/boards/px4/sitl/default.cmake
> -- PX4 config: px4_sitl_default
> -- PX4 platform: posix
> -- PX4 lockstep: enabled
> -- cmake build type: RelWithDebInfo
> -- The CXX compiler identification is AppleClang 11.0.3.11030032
> -- The C compiler identification is AppleClang 11.0.3.11030032
> -- The ASM compiler identification is Clang
> -- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Building for code coverage
> -- ccache enabled (export CCACHE_DISABLE=1 to disable)
> -- Found PythonInterp: /Applications/anaconda3/bin/python3 (found suitable version "3.7.7", minimum required is "3") 
> -- build type is RelWithDebInfo
> -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-384-g77b1112
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /Users/ys/Zeng_Lab/Tailsitter/Firmware/build/px4_sitl_default
> [7/742] git submodule src/drivers/gps/devices
> [12/742] git submodule Tools/sitl_gazebo
> [13/742] git submodule src/lib/ecl
> [18/742] git submodule mavlink/include/mavlink/v2.0
> [19/742] Performing configure step for 'sitl_gazebo'
> -- install-prefix: /usr/local
> -- cmake build type: RelWithDebInfo
> -- The C compiler identification is AppleClang 11.0.3.11030032
> -- The CXX compiler identification is AppleClang 11.0.3.11030032
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Performing Test COMPILER_SUPPORTS_CXX17
> -- Performing Test COMPILER_SUPPORTS_CXX17 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX14
> -- Performing Test COMPILER_SUPPORTS_CXX14 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX11
> -- Performing Test COMPILER_SUPPORTS_CXX11 - Success
> -- Performing Test COMPILER_SUPPORTS_CXX0X
> -- Performing Test COMPILER_SUPPORTS_CXX0X - Success
> -- Using C++17 standard
> -- Found Boost: /usr/local/lib/cmake/Boost-1.73.0/BoostConfig.cmake (found suitable version "1.73.0", minimum required is "1.58") found components: system thread filesystem 
> CMake Warning (dev) at /usr/local/Cellar/cmake/3.18.1/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
>   The package name passed to `find_package_handle_standard_args` (PkgConfig)
>   does not match the name of the calling package (gazebo).  This can lead to
>   problems in calling code that expects `find_package` result variables
>   (e.g., `_FOUND`) to follow a certain pattern.
> Call Stack (most recent call first):
>   /usr/local/Cellar/cmake/3.18.1/share/cmake/Modules/FindPkgConfig.cmake:59 (find_package_handle_standard_args)
>   /usr/local/lib/cmake/gazebo/gazebo-config.cmake:30 (include)
>   CMakeLists.txt:46 (find_package)
> This warning is for project developers.  Use -Wno-dev to suppress it.
[... 251 lines cut ...]
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/unordered_map:407:
> In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__hash_table:18:
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:325:9: error: no member named 'isless' in the global namespace
> using ::isless;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:326:9: error: no member named 'islessequal' in the global namespace
> using ::islessequal;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:327:9: error: no member named 'islessgreater' in the global namespace
> using ::islessgreater;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:328:9: error: no member named 'isunordered' in the global namespace
> using ::isunordered;
>       ~~^
> /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cmath:329:9: error: no member named 'isunordered' in the global namespace
> using ::isunordered;
>       ~~^
> 13 errors generated.
> ninja: build stopped: subcommand failed.
> FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
> cd /Users/ys/Zeng_Lab/Tailsitter/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.18.1/bin/cmake --build /Users/ys/Zeng_Lab/Tailsitter/Firmware/build/px4_sitl_default/build_gazebo -- -j2
> ninja: build stopped: subcommand failed.
> make: *** [Makefile:226: px4_sitl] Error 1

it seems to give the similar error report.

julianoes, 2020-08-05:
Right, I just needed to see which compiler it selected, etc.

I found these:
https://stackoverflow.com/questions/58628377/catalina-c-using-cmath-headers-yield-error-no-member-named-signbit-in-th
https://stackoverflow.com/questions/58313047/cannot-compile-r-packages-with-c-code-after-updating-to-macos-catalina

I'd try to reinstall the XCode Tools: `xcode-select --install`.

ghostSAS, 2020-08-05:
Thanks, Julian!  but still no good luck. I first tried `xcode-select --install`, which leads to 

> xcode-select: error: command line tools are already installed, use "Software Update" to install updates

Then I navigated to your links, but their answer doesn't fix my issue. Even the hard code `cmath` file. Really don't know why. Later I found it is probably because my current compiler cannot find the correct library. As suggested in:
https://github.com/nlohmann/json/issues/2087
https://stackoverflow.com/questions/47018913/make-fails-on-mac
they use `clang++` in the command line as the compiler. In my case, how should I realize the same purpose?

closed 2020-12-18 by Jaeyoung-Lim