PX4/PX4-Autopilot #15484: Compass Inconsistant : Pixhawk4 + Here2

FlavioTonelli, 2020-08-04:
It looks like something break during the last 1 or 2 months on the compass driver/calibration.
No matter how many time I try to recalibrate the compass I always meet the compass inconsistent error when I try to arm the drone.
Tested on Pixhawk 4 + Here 2 gps antenna.
Here are some logs from boot to shutdown during two different calibration attempts:
https://review.px4.io/plot_app?log=82ca641f-166c-4e86-8080-1211799d6af0
https://review.px4.io/plot_app?log=59e34c45-1499-4092-b533-e50cfa0fcee7
https://review.px4.io/plot_app?log=28cdaabc-6c03-4015-9861-32adb561c7b6
https://review.px4.io/plot_app?log=cd98260a-2df1-48e2-8735-3c55b901525f

FlavioTonelli, 2020-08-04:
Could be related to #15312

CornerOfSkyline, 2020-08-11:
@FlavioTonelli @dagar 
I met this error, too. I found the reason is after recalibrate, the sensor_selection could select the compass - compass 1. After reboot the Pixhawk , i found there is no CAL_MAG1_XOFF、CAL_MAG1_XSCALE、CAL_MAG1_YOFF、CAL_MAG1_YSCALE、CAL_MAG1_ZOFF、CAL_MAG1_ZSCALE (they really be caculated by the calibrate, but not be used, just after recalibrate, we can found them)，and sensor_selection could not select the compass. When i trige a param update event, then the CAL_MAG1* be used, and the sensor_selection could select the compass -compass 1. I think the problem is the Here+ compass 1 update mag data too slow, missing the sensors update phase in the boot logic.

I think the problem is here.
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L158

nrogelio, 2020-08-24:
Hi,

A PR was merged to master: #15235

You can try to write master firmware to your board and see if that solves the issue. Of course master firmware is a developer version so use it at your own risk.

Regards.

FlavioTonelli, 2020-08-31:
Hi @nrogelio, I just tried a calibration with the last master and I'm right now able to arm the drone.
Here are the logs of the working test:
During calibration: https://review.px4.io/plot_app?log=e1decf84-6bdc-422b-8acf-70a3a4e8f701
After calibration: https://review.px4.io/plot_app?log=477a467e-2cbc-40ee-9789-07157729ed3b

@dagar may I ask the reason why https://github.com/PX4/Firmware/pull/15235 is not included in the upcoming release? This is preventing the future stable to work with my hw configuration.

dagar, 2020-09-01:
> @dagar may I ask the reason why #15235 is not included in the upcoming release? This is preventing the future stable to work with my hw configuration.

Because it's a huge change that needs time for testing. The intention is that it fixes and improves things, but there's always the potential for a regression.

For right now based on your logs you can simply increase `COM_ARM_MAG_ANG` to 45 degrees, which is one of the (many) changes in https://github.com/PX4/Firmware/pull/15235.

kuiskas, 2020-10-12:
Hi, 
same issue here.
Is there any workaround to use this hardware and the the version 1.11.1? What should I take in consideration?
Alternatively, which RTK do you recommend for the Pixhawk 4?

kuiskas, 2020-10-12:
Is the new Holybro RTK plug & play?