PX4/PX4-Autopilot #15490: BMI088 driver cannot be built on posix platform now

UAV-Pilot, 2020-08-05:
**Describe the bug**
BMI088 driver is only usable on nuttx platform now. It worked before on posix/linux platform, but now it cannot be built for posix platform in the master branch due to the following:

 #include <px4_platform/board_dma_alloc.h>

File px4_platform/board_dma_alloc.h is only available on nuttx platform at this point. The change/improvement for this drive on nuttx platform should not break it on posix platform.

**To Reproduce**
Steps to reproduce the behavior:
1. add the following driver to cmake file of a posix board:
     imu/bosch/bmi088
2. the following build error comes up:

Firmware/src/drivers/imu/bosch/bmi088/BMI088_Gyroscope.cpp:36:10: fatal error: px4_platform/board_dma_alloc.h: No such file or directory
 #include <px4_platform/board_dma_alloc.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

**Expected behavior**
BMI088 driver can be built and continues to work on posix platform as before.

julianoes, 2020-08-05:
@UAV-Pilot which platform are you actually using?

UAV-Pilot, 2020-08-05:
I'm using NVidia Jetson Nano, but the problem should exist on any posix/linux platform as file "px4_platform/board_dma_alloc.h" only exists for nuttx.

slgrobotics, 2024-05-16:
The following PR will remove the offending ```#include <px4_platform/board_dma_alloc.h>```:

https://github.com/PX4/PX4-Autopilot/pull/23141

I build this code on an Intel/Ubuntu 22.04 machine for Raspberry Pi 4 (arm64) - a Posix system.