PX4/PX4-Autopilot #15497: VTOL will not convert to FW when motors run

bukovec1, 2020-08-05:
Hello,

All tests were done on the bench. 

We manually set a switch on the transmitter that would allow us to "transition" from MC to FW. We could arm the ship and **without any motors spinning**, I could flip the switch and QGC would show the vehicle would go into FW state. 

Unfortunately, when the motors are spinning and I flip the switch to put the vehicle in FW mode, nothing happens. QGC does not show that the ship is in FW, it just stays constant in MC mode. 

I am wondering if anyone else has had this issue with 1.11.

dagar, 2020-08-05:
Can you share the log (https://logs.px4.io/)? I'm wondering if there's an interaction with the land detector you're seeing.

bukovec1, 2020-08-05:
Hello,

Can you view this file from this url?

https://review.px4.io/plot_app?log=d5c46412-2c4c-4447-83ba-bb668a674d89

notoriou5, 2020-08-06:
@bukovec1 I experienced the same issue, but in the opposite direction. Once the motors where off, I could not do the transition. Did you connect your tilt servos through the AUX Channel? I changed PWM_AUX_DISARMED and PWM_AUX_MIN both to 1000 and it worked again. Not sure whats the reason for that. Hopefully it helps you as well

bukovec1, 2020-08-06:
@notoriou5 

Thanks for your comment. Unfortunately 1.11 does not have  PWM_AUX_DISARMED and PWM_AUX_MIN exactly, but I did try to change both PWM_MIN and PWM_DISARM to 1000 and neither solved my issue yet.

xdwgood, 2020-08-11:
I have same problem.(on two motor tailsitter)

fix by here:
https://github.com/PX4/Firmware/pull/15493
https://github.com/PX4/Firmware/pull/15352

bukovec1, 2020-08-11:
@dagar is this solution valid? I was looking at #15493 and it still looks like it needs approved by someone? 

@xdwgood thank you for this. I will try it to see what happens

xdwgood, 2020-08-12:
@bukovec1  If you tested with #15493 , it would be great to provide feedback

bukovec1, 2020-08-19:
Hello @xdwgood  @dagar @sfuhrer @notoriou5 

I found the issue. It was because the transition airspeed was set to 10 m/s, and since I was inside bench testing, this was not happening. When setting this to 0, I can transition.

sfuhrer, 2020-08-25:
> Hello @xdwgood @dagar @sfuhrer @notoriou5
> 
> I found the issue. It was because the transition airspeed was set to 10 m/s, and since I was inside bench testing, this was not happening. When setting this to 0, I can transition.

Okay, then I guess we can close this issue?
You can also blow briefly into the pitot tube when doing transitions on a bench setup to fake airspeed.

xdwgood, 2020-08-26:
@sfuhrer  As this log shows, the flight transmission failed because there was no pitch setting value, not because the conversion airspeed was not reached.https://github.com/PX4/Firmware/issues/15497#issuecomment-671822763(see picture)

I hope it will be completely resolved, rather than hidden.I am not against closing it, but we should continue to work here:https://github.com/PX4/Firmware/pull/15493

sfuhrer, 2020-08-26:
@xdwgood this log file:

> https://review.px4.io/plot_app?log=d5c46412-2c4c-4447-83ba-bb668a674d89

is from a standard VTOL, and your issue is with a tailsitter right? So I don't think they're related.

xdwgood, 2020-08-27:
@sfuhrer  I remember that `VTOL Quad Tailsitter (13003)` called tailsitter.cpp

`VTOL Quad Tailsitter`  not `standard VTOL`

You can check the code again to verify this

[image]

vtol_type = 0 means:

[image]

yuthikasagarage, 2020-10-03:
in the tailsitter also, if you blow across the pitot, it transitions to fw. and after the normalization was added --->

https://github.com/PX4/Firmware/blob/8ee0c62e575435f4bb751599ae277bab1208c809/src/modules/vtol_att_control/tailsitter.cpp#L216

i didn't come across any pitch setpoint publishing error.

sfuhrer, 2021-01-18:
Closing as the cause for the original issue was found 
>I found the issue. It was because the transition airspeed was set to 10 m/s, and since I was inside bench testing, this was not happening. When setting this to 0, I can transition.

Please open a new issue if still present on some setups.

closed 2021-01-18 by sfuhrer