PX4/PX4-Autopilot #15505: SITL Iris: `BAT_A_PER_V`, 'BAT_V_DIV' missing from parameter list

DonLakeFlyer, 2020-08-06:
* Sync to latest firmware laster
* `PX4_SIM_SPEED_FACTOR=2 HEADLESS=1 make px4_sitl_default gazebo_iris`
* Start QGC, go to Power page
* It will report the two param as missing. They are required by QGC power setup to exist.
```
FactPanelControllerLog: Missing parameter: "1:BAT_A_PER_V"
FactPanelControllerLog: Missing parameter: "1:BAT_V_DIV"
```

DonLakeFlyer, 2020-08-06:
Problems seems to be SITL only. Flash master to a board and the params are there.

dagar, 2020-08-06:
Thanks for reporting this Don. Those parameters aren't there because SITL doesn't have analog power input in any form. I can rearrange things to keep it in the build for now.

At a minimum could we maintain a list of parameters QGC expects to exist? Then for optional things we could talk about the appropriate mechanisms to detect.

DonLakeFlyer, 2020-08-06:
> At a minimum could we maintain a list of parameters QGC expects to exist? Then for optional things we could talk about the appropriate mechanisms to detect.

That's not super easy to do as it currently stands. It just happens randomly. Let me figure out how to re-architect QGC ui so that this becomes obvious. Not going to happen super quick.

DonLakeFlyer, 2020-08-06:
I think there is a new stable in RC? If so, this is important for that.

bys1123, 2020-08-06:
`make px4_sitl_nolockstep flightgear_rascal`
will shows sensors not calibration and BAT_* parameters missing

modaltb, 2020-08-06:
@dagar please let me know if this same thing can be applied to what I think is holding up https://github.com/PX4/Firmware/pull/14852
thanks!

dagar, 2020-08-12:
Should be fixed by https://github.com/PX4/Firmware/pull/15535.

closed 2020-08-12 by dagar