PX4/PX4-Autopilot #15506: MAV_SYS_STATUS_PREARM_CHECK doesn't seem to indicate ready to fly correctly

DonLakeFlyer, 2020-08-06:
I'm trying to use this as a generic "Ready To Fly" indicator.

* Firmware master
* PX4_SIM_SPEED_FACTOR=2 HEADLESS=1 make px4_sitl_default gazebo_iris
* Add the following code to QGC handler for sys_status
```
    qDebug() << (sysStatus.onboard_control_sensors_present & MAV_SYS_STATUS_PREARM_CHECK ? "PreArm Check Available" : "PreArm Check NOT Available")
             << (sysStatus.onboard_control_sensors_enabled & MAV_SYS_STATUS_PREARM_CHECK ? "PreArm Check enabled" : "PreArm Check disabled")
             << (sysStatus.onboard_control_sensors_health & MAV_SYS_STATUS_PREARM_CHECK ? "PreArm READY" : "PreArm NOT ready");
```
* As soon as SITL vehicle boots QGC immediately shows `PreArm Check Available PreArm Check enabled PreArm READY`. I thought SITL went through a GPS 3D lock delay to simulate GPS acquisition. PreArm shouldn't be ready until 3d lock is acquired. If SITL doesn't have some sort of GPS acquisition delay it should so it can be used to test these situations.
* If I upload a mission which the first waypoint as being too far away. PreArm check still comes back as ok. Even though you can't fly.

DonLakeFlyer, 2020-08-06:
I think there is a new stable in RC? If so, this is important for that.

Antiheavy, 2020-08-08:
The parameter COM_ARM_MIS_REQ requires a "valid mission to arm".  Does this parameter correctly influence the preArm check?  Seems like COM_ARM_MIS_REQ should cause the preArm check to look at MAV_MISSION_RESULT?  To be honest I can't quite follow the code for how all this ties together, but there seems to be some missing status.

DonLakeFlyer, 2020-08-08:
Ah, didn't know that existed. If you turn that it fails pre-arm. I would content that the default for COM_ARM_MIS_REQ should be true. Also the meta data for the parameter should specify an enabled/disabled enum values so you get good ui to set it.

DonLakeFlyer, 2020-08-14:
Wait, I'm reading this wrong. That parameters control whether a mission is required to fly. It should not also control whether a bad mission fails the pre-arm check if that is set to off. To me that should alway fail pre-arm.

Antiheavy, 2020-08-14:
>  It should not also control whether a bad mission fails the pre-arm check if that is set to off. To me that should alway fail pre-arm

I agree, but I wonder if others would argue other ways?  this would mean nobody could arm if there was some old mission onboard the vehicle, even if they never plan to use the mission on that flight.  Personally I think an old mission from some other location has no business being onboard the autopilot so it should be forced to be deleted before arming, but I bet you there are people out there who will argue otherwise....