PX4/PX4-Autopilot #15508: Wind estimator don't compute sideslip

moreba1, 2020-08-07:
Hi

In my recent test, i see 30-degree sideslip angle during cruise with about 19 m/s airspeed but wind speed is 3.7 m/s ,
estimated wind must be about 9.5 m/s.

https://review.px4.io/plot_app?log=b7e0e58f-ee70-423c-a51e-3c583a4eb009

[image]

bresch, 2020-08-07:
FYI @RomanBapst

bresch, 2020-08-07:
@moreba1 I don't think this is a wind estimator issue, the sideslip you see does not seem to be caused by the crosswind: the ground and airspeed have similar values during the whole survey, this is impossible with a wind of 9m/s.
[image]

moreba1, 2020-08-07:
> @moreba1 I don't think this is a wind estimator issue, the sideslip you see does not seem to be caused by the crosswind: the ground and airspeed have similar values during the whole survey, this is impossible with a wind of 9m/s.

airspeed and ground speed can be the same value but we have wind speed because they are vector.

[image]

Jaeyoung-Lim, 2020-08-10:
@moreba1 which release are you running? previously airspeed sensor was not simulating effects of wind. 

Also, some relevant issues from @ThomasRigi in https://github.com/PX4/sitl_gazebo/pull/557

moreba1, 2020-08-10:
> @moreba1 which release are you running? previously airspeed sensor was not simulating effects of wind.

version 1.10 latest stable version

bresch, 2020-08-10:
> Airspeed and ground speed can be the same value but we have wind speed because they are vector.

Yes, but it your mission, you flew in all directions and none of them is showing a difference of more than 4-5 m/s between ground and air speed. Are you sure that the plane was really crabbing?

Furthermore, if you check the 3d replay of your log (https://review.px4.io/3d?log=b7e0e58f-ee70-423c-a51e-3c583a4eb009), you can see that the yaw offset isn't consistent when going N-W and S-E:
[image]
[image]
So we have heading issues and NOT wind estimator issues.

bresch, 2020-08-10:
@moreba1 Closing as this is most likely an issue with your setup and not the firmware. Please go to http://discuss.px4.io and if you find that the issue is not from you setup but from the estimator itself, please report to https://github.com/PX4/ecl

closed 2020-08-10 by bresch

moreba1, 2020-08-10:
> Yes, but it your mission, you flew in all directions and none of them is showing a difference of more than 4-5 m/s between ground and air speed. Are you sure that the plane was really crabbing? [...]

does not wind estimator use heading of the plane ? so how sideslip angle calculated?