PX4/PX4-Autopilot #15513: Cube Yellow error "[IO] config read fail, abort."

60ericleal60, 2020-08-07:
I have flashed px4 1.11.0rc to my cube yellow on Ubuntu 18.04 and an getting the error "[IO] config read fail, abort." There also seems to be issues like the imu heater not turning off and the Frsky radio not being read. The board is still not recognized by QGC and I understand this is still a work in progress.

dagar, 2020-08-07:
Can you try current master? This should already be fixed for the next RC. The target is now `cubepilot_cubeyellow_default`.

60ericleal60, 2020-08-07:
Oh thanks. I just recloned the repository and those issues went away. The described above error is from cloning the master on Monday.

dagar, 2020-08-07:
Thanks for the update, it sounds like https://github.com/PX4/Firmware/pull/15415 fixed it.

For flashing with QGC we can quickly fix that by manually adding the Cubes like this. https://github.com/mavlink/qgroundcontrol/pull/8929

A better solution is to get all the firmware into a json manifest.

731avenger, 2020-08-07:
I do not know if it changes anything but QGC is already seeing the bootloader but previously was not able to flash an image as none existed. #14924

60ericleal60, 2020-08-10:
I may have been premature in thinking my problem was fixed. The second time I powered on the flight controller I got this warning again. I also tried the master from today.

dagar, 2020-08-11:
> I may have been premature in thinking my problem was fixed. The second time I powered on the flight controller I got this warning again. I also tried the master from today.

@60ericleal60 can you confirm if the IO is actually working the rest of the time?

60ericleal60, 2020-08-11:
I downloaded the .px4 file from the master today and flashed from QGC the custom firmware file and am still getting the error message on boot. I have also built the code on my Ubuntu machine. The first obvious setup error comes when I see no channels on my frsky pilot radio when it works with the ardupilot firmware.

Is there a way I can share with you a more verbose boot message log or other information I am using?

731avenger, 2020-08-11:
I also have been trying to get the PX4 on to the Cube Yellow and I am getting the exact same results as 60ericleal60. I have tried using the PX4 files from today, from https://github.com/PX4/Firmware/pull/15415/checks and before the pull request. All have the "[IO] config read fail, abort." I have not tried to build the code but only using PX4 generated files.

60ericleal60, 2020-08-12:
I have found maybe an additional clue. When I disconnect the serial GPS unit (I have tried Holybro and Here+) and reboot the flight controller this [IO] config read fail error does not appear. I can connect to the flight controller through both usb and sik radio and all of the sensors work including the external mag with or without the GPS plugged in. When the GPS is plugged in I see no satellites and I do not get a GPS mavlink message. In both cases the sbus pilot radio does not work.

dagar, 2020-08-16:
I have some small fixes for the Cube yellow, but I'm still seeing px4io issues.

``` Console
ERROR [px4io] config read error
ERROR [px4io] [IO] config read fail, abort.
ERROR [px4io] driver init failed
WARN  [px4io] px4io is not started, still attempting upgrade
[PX4IO] using firmware from /etc/extras/cubepilot_io-v2_default.bin
[PX4IO] bad sync 0xff,0x7f
[PX4IO] bad sync 0xff,0xff
[PX4IO] found unsupported bootloader revision 269619218, exiting

```

dagar, 2020-08-16:
I manually flashed the IO on my Cube Yellow and it's now fully working.

dagar, 2020-08-16:
Everything works if I flash the IO through SWD, but later if if I try to force update the bootloader version check fails.

JacobCrabill, 2020-08-18:
@dagar could it be related to ${IOFW} being hard-coded to `/etc/extras/px4_io-v2-default.bin` in rcS?
That wouldn't explain the post-boot "force load" issue you posted; that shows the correct file...

Did you update the IO bootloader as well with SWD?  Could it be a manifest / load address isssue?

I feel like I've encountered this error before on a Black Cube, but I can't recall what the cause/fix was.

Murcamil, 2020-12-07:
Hi, anything new on this issue?

I have several Cubes Yellow, and, as the OP, I get the "[IO] config read fail, abort." in QGC. 
I have flashed with the latest QGC(v4.0.11) in Windows by selecting a custom firmware because the automatic way failed, with px4 V1.11.2 - cubepilot_cubeyellow_default.px4. Flashing in Ubuntu 18 with the same QGC does not yield a different result. 
I tried powering it either from the USB or a ProfiCNC Power Brick without being able to get this error out of the way.

leekyubum, 2020-12-22:
Hi, 
Is this problem fixed? 
anything new on this issue?

731avenger, 2020-12-25:
I do not believe anything has changed, the wiki still marks this as a supported target but still is not flashable using normal methods.

dagar, 2020-12-25:
Newer QGC should be able to flash it, but as far as I know the px4io issue hasn't been resolved.

leekyubum, 2021-01-05:
Hi,
is there any different io-v2 and cubepilot_io-v2 ?
if there is no difference, Can I set to use io-v2?

/board/cubepilot/default.make  # file directory
line 10
-------------------------
IO cubepilot_io-v2_default to IO px4_io-v2_default
-------------------------

leekyubum, 2021-01-05:
test

leekyubum, 2021-01-05:
Hi,
is there any different io-v2 and cubepilot_io-v2?
if there is no difference, Can I set to use io-v2?

/board/cubepilot/default.make # file directory
line 10
IO cubepilot_io-v2_default to IO px4_io-v2_default

SPuigUAVW, 2023-11-07:
I think this issue still unresolved. Today I have tryied to configure a Cube Yellow and I haven't been capable to read any SBUS/PPM radio signal and I have not checked with the osciloscope, but I think that no signal is beeing outputing to the motor pins.

I checked with the console "px4io" and the service wasn't running and when I tried to start it, same error popped "[IO] config read fail, abort."

julianoes, 2023-11-10:
@SPuigUAVW can you share the output of `dmesg` (e.g. in QGC mavlink shell) after startup please?