PX4/PX4-Autopilot #15526: question about pwm capture for v5 nano (fmu-v5 1.10.2)

liganggg, 2020-08-10:
hi
when i want to use fmu-v5(version 1.10.2) to capture a edge from other board, 
i had config the param 
CAM_CAP_FBACK = 1
at the start last print in console,i can see 
"
INFO  [camera_capture] Not in a capture mode                                                                                                                                                        
INFO  [camera_capture] Mode changed to 4PWM2CAP 
"
but when i run "fmu test "in console
it print 
"
INFO  [fmu] Testing 4 servos and 2 input captures                                                                                                                                                   
ERROR [fmu] Unable to get capture callback for chan 4 
"

i have some question about it
I'm not sure the configuration is correct; 
and when config 4PWM2CAP is it means that the M1--M4 is pwm out ,and which pin can use for capture edge? AUX1 and AUX2?
 
Any help would be appreciated,
Thanks,

nrogelio, 2020-08-20:
Hi,

https://docs.px4.io/master/en/assembly/quick_start_cuav_v5_nano.html#wiring-chart-overview mentions A1-A3 are capture pins but not currently suppported in PX4.

Also found this post in forums, it's not specific to V5 nano, but you can give it a try: https://discuss.px4.io/t/camera-feedback-wiring-and-setting/13569

Regards.

hamishwillee, 2020-08-24:
@bkueng As you recall we just updated http://docs.px4.io/master/en/peripherals/camera.html#hardware_setup to indicate FMU PWM pin 6 is camera capture on PX4 master, but this does not mention having to set CAM_CAP_FBACK - do you have to set that?

Also, do you you know if the CUAV camera capture pins are supported as per https://github.com/PX4/Firmware/issues/15526#issuecomment-677719956 ?

bkueng, 2020-08-24:
> do you have to set that?

Yes

@liganggg the configuration should be correct. You can check with `camera_capture status` and `listener camera_trigger` for trigger events.

liganggg, 2020-09-03:
> > do you have to set that?
> 
> Yes
> 
> @liganggg the configuration should be correct. You can check with `camera_capture status` and `listener camera_trigger` for trigger events.

as my setting (CAM_CAP_FBACK = 1)
in the code fmu.cpp it execute 
		up_input_capture_set(2, Rising, 0, NULL, NULL);
		up_input_capture_set(3, Rising, 0, NULL, NULL);

it mean that set FMU3(PE11) FMU4(PE9), 
but nano has no io board, so it pmu1--8 map the M1--8, 
for hw with  px4io board(stm32f100) ,pmu1--8 map the aux1--8,
so i still can not make the capture function work well,
even use command camera_capture status it get all zero like this
INFO  [camera_capture] Capture enabled : YES
INFO  [camera_capture] Frame sequence : 0
INFO  [camera_capture] Last trigger timestamp : 0
INFO  [camera_capture] Number of overflows : 0

the same setting at cuav v5+ work well 
so i think it dont support the capture funtion without px4io board.
any suggestion?  thanks