PX4/PX4-Autopilot #15527: MRO X2.1-777

taileron, 2020-08-10:
**Describe the bug**
IO doesn´t start and gets no firmware

**To Reproduce**
Steps to reproduce the behavior:
#15521 solved perfectly the issues about booting and usb-mavlink 
pwm out starts only with sys_use-io = 0

**Expected behavior**
8 pwm out are available with sys_use_io = 1

**Log Files and Screenshots**

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: MRO_X21_777
FW git-hash: 97fc1db768b2ac8fe4121551402912ac57369e1f
FW version: 1.11.0 0 (17498112)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 2a33d2b6dca623556c7276c5d592d9238bdaff52
Build datetime: Aug 10 2020 15:08:44
Build uri: localhost
Toolchain: GNU GCC, 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
PX4GUID: 0002000000003237323134385108002d0022
MCU: STM32F76xxx, rev. Z
ERROR [mtd] failed to initialize mtd driver
INFO  [param] selected parameter default file /fs/microsd/params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
WARN  [px4io] check CRC failed: -22
INFO  [tune_control] Publishing standard tune 18
ERROR [px4io] config read error
ERROR [px4io] [IO] config read fail, abort.
ERROR [px4io] driver init failed
WARN  [px4io] px4io is not started, still attempting upgrade
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] bad sync 0x7f,0x7f
[PX4IO] found unsupported bootloader revision 269619218, exiting
INFO  [tune_control] Stopping playback...
WARN  [px4io] check CRC failed: -1
INFO  [tune_control] Publishing standard tune 20
Board sensors: /etc/init.d/rc.board_sensors
ms5611 #0 on SPI bus 1 (devid=0x3d)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 48000 B/s on /dev/ttyS1 @ 921600B
INFO  [init] Mixer: /fs/microsd/etc/mixers/vtol_convergence.main.mix on /dev/pwm_output0
ERROR [param] Parameter SENS_EN_PX4FLOW not found
Addons script: /fs/microsd/etc/extras.txt
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero
INFO  [ecl/EKF] 14813659: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [mavlink] Starting mavlink shell

nsh> px4io start
ERROR [px4io] Failed to communicate with IO, abort.
ERROR [px4io] driver init failed

dagar, 2020-08-10:
Thanks for the quick update, I'll take a look at this next.

dagar, 2020-08-10:
@taileron do you have the safety button connected? If so, please try powering on the board with the safety button held to force the px4io upgrade.

taileron, 2020-08-10:
Pins are not yet soldered but I can manually connect the contacts and this message is a little more specific so I guess that even a forced firmware update does not work: 
px4io update
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] found unsupported bootloader revision 269619218, exiting
nsh> 

The safety button didn´t work as well:
ERROR [mtd] failed to initialize mtd driver
INFO  [param] selected parameter default file /fs/microsd/params
INFO  [tune_control] Publishing standard tune 1
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 362560 bytes
WARN  [px4io] check CRC failed: -22
INFO  [tune_control] Publishing standard tune 18
ERROR [px4io] config read error
ERROR [px4io] [IO] config read fail, abort.
ERROR [px4io] driver init failed
WARN  [px4io] px4io is not started, still attempting upgrade
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] bad sync 0x7f,0x7f
[PX4IO] bad sync 0xff,0xff
[PX4IO] found unsupported bootloader revision 269619218, exiting
INFO  [tune_control] Stopping playback...
WARN  [px4io] check CRC failed: -1
INFO  [tune_control] Publishing standard tune 20
INFO  [tune_control] Publishing standard tune 2
Board sensors: /etc/init.d/rc.board_sensors
ms5611 #0 on SPI bus 1 (devid=0x3d)
icm20602 #0 on SPI bus 1 (devid=0x38)
mpu9250 #0 on SPI bus 1 (devid=0x24)
Board extras: /etc/init.d/rc.board_mavlink
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 48000 B/s on /dev/ttyS1 @ 921600B
ERROR [mixer] can't open /dev/pwm_output0

ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /fs/microsd/etc/mixers/vtol_convergence.main.mix
INFO  [tune_control] Publishing standard tune 20
ERROR [vtol_att_control] can't open /dev/pwm_output0
ERROR [param] Parameter SENS_EN_PX4FLOW not found
Addons script: /fs/microsd/etc/extras.txt
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
ERROR [pwm] can't open /dev/pwm_output0
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [ecl/EKF] reset position to last known position
INFO  [ecl/EKF] reset velocity to zero

taileron, 2020-08-12:
MTD cannot be activated yet , as it is not absolutely necessary if an SD is available. Should I try to find out if the M25V02 on the F777 is connected to the same SPI pins as the X2.1 for completeness, or is there an exact schematic for X2.1-777 available?

taileron, 2020-10-05:
Tried to flash io with St-Link V2 by SWD. Which bin file is the right one? As soon as px4_io-v2.bin from build/external/build is flashed the whole fc doesn´t boot any more it only starts to boot again if a standard px4io_bl.bin bootloader is flashed afterwards. But still no FC communication to IO.

taileron, 2020-10-06:
IO now works with the master from 5´th of september flashed by SWD ... further tests will follow ...

dagar, 2020-10-06:
Thanks for the update.

taileron, 2020-10-06:
8 PWM outs and wifi telemetry 1 are now working but after each changed main firmware the io has additional to be flashed by SWD (pressing safety sw doesn´t force io flash at boot).
standard px4io_bl bootloader, master from august 2020 was flashed before. 
QGC allows only to flash PX4 firmware from custom file.

taileron, 2022-01-11:
can be closed only #17558 is left for this board

closed 2022-01-11 by taileron