PX4/PX4-Autopilot #15532: ESC calibration not working correctly when using a custom airframe

Spir4L, 2020-08-11:
**Describe the bug**

If I setup PX4 to use a custom airframe in order to use a non standard setup, the ESC calibration done with QGroundControl doesn't work correctly, i.e. the ESCs max values seem to be set following the airframe mixing strategy (if the mixer is not splitting the thrust equally among the different motors). This results in the aircraft motors saturating earlier then expected.
In order to fix the issue I have to set a standard airframe before calibration (e.g. Hexa X) and then set the new airframe after calibration. 

I think this is happening because the calibration is done setting a thrust requirement that is mixed into a PWM requirement, instead of directly PWM requirements.

**To Reproduce**
Create an airframe in which the thrust is not split equally between the motors, and try calibrating the ESCs with QGroundControl. Look at the ESC calibration values.

**Expected behavior**
I would expect the ESC calibration to be completely uncoupled from the actual airframe selected, and to be tuned based on their own max and min values.

**Log Files and Screenshots**
I am not sure how I could show this, as I am using a custom airframe that I have set for my

dagar, 2020-08-12:
I believe one of the current impediments to doing it directly based on PWM_MIN/PWM_MAX is not knowing which outputs are actually ESCs vs servos.

Spir4L, 2020-08-13:
Thanks @dagar, I think this should be at least documented, or maybe there is a mitigation that can be found?