PX4/PX4-Autopilot #15541: Bug: Driver DPS310 does not pass the I2C address to I2CSPIDriverBase

Ecuashungo, 2020-08-13:
**Describe the bug**
A clear and concise description of the bug.
I connected the control zero flight controller from mrobotics and flashed it with the latest px4 Firmware (see screenshot).

In the dmesg of the mavlink console it states that there is a driver issue for the barometer dps310.

[image]

How can this be fixed?
Also QGC doesn't detect the radio at all, wether I use PWM with S.BUS or PPM. Could this be related to the driver issue?

dagar, 2020-08-13:
I believe it's just a warning, although I'd still like to fix it.

You can check the output of `sensors status` or `listener sensor_baro` to see if you have barometer data.

dagar, 2020-08-13:
Could you give https://github.com/PX4/Firmware/pull/15215 a quick try and tell me if RC works?

Ecuashungo, 2020-08-13:
Both commands showed a working baro, according to my understanding. I attached the screenshots for reference.

> I believe it's just a warning, although I'd still like to fix it.
> 
> You can check the output of `sensors status` or `listener sensor_baro` to see if you have barometer data.

`listener sensor_baro`
[image]

`sensors status`
[image]
Why is this command showing an airspeed sensor when I don't have any connected?

-- -- -- 

I have flashed https://github.com/PX4/Firmware/pull/15215 (branch pr-mro_ctrl_zero_power_init) onto the control zero, but the RC calibration gives the same error (Detected 0 radio channels. To operate PX4, you need at least 5 channels.)

The dmesg output for this try:
[image]