PX4/PX4-Autopilot #15557: Support gyros/accelerometers with multiple separate SPI devices

lukegluke, 2020-08-16:
For now (as I see) architecture of IMU drivers in PX4 proceeds from the assumption that an accelerometer or gyroscope is a single 3 axis device.
But there is a pool of high stability sensors that use separate chips for different axes, for example muRata SCC2000 Series (for instance IMU based on two [SCC2130](https://www.murata.com/en-global/products/sensor/guide/sensorguide3/sensorguide/scc2130-d08) gyro for X and Y and [SCC2230](https://www.murata.com/en-global/products/sensor/guide/sensorguide3/sensorguide/scc2230-d08) for Z).
Also in addition those gyros contains 3-axis accelerometer each.

dagar, 2020-09-08:
Sorry, I thought I responded to this already. Short answer is yes we can support this if necessary. 

The first idea that comes to mind is to write separate drivers for the SCC2130 and SCC2230, but then have a wrapper that handles them together to publish a single `sensor_gyro`.

We could also consider separately publishing gyro data per axis and piecing it together in the sensors module, but I don't think it's worth it unless the use case becomes common.

lukegluke, 2020-09-08:
@dagar thanks for feedback!

closed 2020-09-08 by lukegluke