PX4/PX4-Autopilot #15559: Matek systems optical flow integration with PX4 firmware

Chick92, 2020-08-16:
Hi, 

I'm aware that the sensors on this board are individually supported by the PX4, however, would it be possible to integrate the board as a whole, as has been done with iNAV? It uses the MSP protocol. 

http://www.mateksys.com/?portfolio=3901-l0x

http://www.mateksys.com/downloads/setup_optic%20flow_in_INAV_rev1.pdf

dagar, 2020-09-08:
Yes it would be possible, and likely not too hard from what I can see. I suppose one question is if there's value in supporting yet another interface to the same sensor, but I'm not opposed to it.

If anyone with one of these devices is interested in writing a driver I'd be happy to help point them in the right direction.

pcguglielmino, 2020-11-26:
Hi,

I currently found myself with one of these Matek sensors and I would like to use it with my PixHawk 4 Mini. Do you think you could help point me in the right direction to write a drive @dagar?

r1b4z01d, 2022-02-16:
any update? I'd be interested in being pointed in the right direction.

BA-ED, 2022-06-23:
I'm also interested in this. I had a look at the serial data sent from the sensor and it's MSP v2 data. Do we have a driver for this at all?

Seeelefant, 2024-03-31:
Is there any news on this topic?

fastworks, 2024-10-19:
I have the same issue. Not sure if I will get around to implementing the MSP protocol. It seems that it is used by a number of sensors, so there could be value in writing a driver for it.