PX4/PX4-Autopilot #15564: DShot module motor count gives 0

Safranek42, 2020-08-17:
**Describe the bug**
When running the DShot module the dshot telemetry should go through all the connected motors. I have one motor connected on AUX3 port, and I can drive it with virtual joystick from QGroundControl. However telemetry does not work, because the dshot module does not recognize that the motor is there, i.e. the `num_motor` variable in src/drivers/dshot/dshot.cpp `updateTelemetryNumMotors()` functions remains 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone 1.11.0 release candidate number 2
2. Use SYS_AUTOSTART 4100 (Tilt_Quadrotor) airframe.
3. Set Dshot mode as described [here](https://docs.px4.io/master/en/peripherals/dshot.html).
4. In QGC mavlink consol give command dshot telemetry /dev/ttyS2 (or where the dshot telemetry cable is connected) to initialize telemetry.
5. I have put `PX4_INFO("number of motors: %d", motor_count);` in updateTelemetryNumMotors() function to print it to the log file.

**Expected behavior**
Number of motors printed to log file should be 1.

**Log Files and Screenshots**
LOG file
https://logs.px4.io/plot_app?log=4d535223-5f86-4189-8fa2-884987608c89

[image]

**Drone (please complete the following information):**
- Describe the type of drone.
Tilt-Quadrotor, `SYS_AUTOSTART=4100`
- Photo of the IMU / autopilot setup if possible.
Flight controller board Holybro Durandal, PX4 1.11.0 second release candidate (https://github.com/PX4/Firmware/releases/tag/v1.11.0-rc2)

**Additional context**

mhkabir, 2020-08-17:
@bkueng

bkueng, 2020-08-24:
What is the output of `dshot status`? Which ESC's do you use? Are you sure nothing else runs on that UART?

Safranek42, 2020-08-25:
Thanks for the reply.
This is the output I get for dshot status (see image).
The type of esc is also there on the screenshot: TEKKO32_F3_Metal

There are only one GPS and the esc with the telemetry wire is connected to my board at the moment. so nothing else should occupy that port. The `dshot status` and the first `dshot esc_info -m 3` command was sent when the FC was armed and the motor was spinning.

On the second image there is the output in the log file.

(I was able to solve the issue with setting the motor_count parameter to one in the code, but I know this is not the most appropriate solution.)

[image]

[image]

bkueng, 2020-08-27:
So is there anything we should fix?

Safranek42, 2020-08-28:
I think it should definitely work with the firmware detecting the motors, hard coding the number of motors for every setup is not the correct way to solve it.

closed 2020-12-03 by Safranek42

Safranek42, 2020-12-03:
Although auto-detecting the number of motors would be a great feature, closing this issue as hardcoding the number of motors in /src/drivers/dshot/dshot.cpp with in `updateTelemetryNumMotors()` function using the `motor_count` variable is the present solution.