PX4/PX4-Autopilot #15570: Missing board definition for CUAV X7 (non-Pro)

thopiekar, 2020-08-18:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
Missing board definition for the device mentioned.

**Describe your preferred solution**
Being able to build PX4 for the device mentioned.

**Describe possible alternatives**
Is the CUAV Nora definition working with the CUAV X7, too?

**Additional context**
-

hamishwillee, 2020-08-19:
If it is PX4 standard (ie. Pixhawk compatible) then it won't have a specific board definition and would be under https://github.com/PX4/Firmware/tree/master/boards/px4.

If it is supported then it will be under https://github.com/PX4/Firmware/tree/master/boards/cuav/x7pro

thopiekar, 2020-08-26:
The X7 is basically a Nora, but with damped sensors in the cube.

closed 2020-08-26 by thopiekar