PX4/PX4-Autopilot #15580: FMU is sending no UDP telemetry stream

dk7xe, 2020-08-19:
**Describe the bug**
PX4 is not sending a UDP telemetry stream via ethernet on FMUK66.
Telemetry stream via UDP can only be enabled if other device (e.g. QGC) starts the connection towards FMU.

https://github.com/PX4/Firmware/issues/15036#issuecomment-646985999 and later

**To Reproduce**
Steps to reproduce the behavior:
1. adjust /etc/extras.txt on SDcard of FMUK66 with "mavlink start -x -u 14550 -r 200000 -t 10.0.0.5 -m onboard" where -t flag defines IP address of computer running QGC.
2. connect ethernet cable via T1 media converter to T1 ethernet port on FMUK66
4. reboot FMUK66
5. telnet into FMU (FMUK66 has 10.0.0.2 as default IP) or access PX4 shell via serial
6. issue "mavlink status"
7. mavlink instance via UDP is shown but not actively connected.
8. setup an outgoing UDP connection on QGC towards FMU. reference described on https://github.com/PX4/Firmware/issues/15036#issuecomment-646985999 
9. QGC receives data from FMU after connect via UDP.

**Expected behavior**
Expectation is that FMU is starting an UDP communication to another device IP/UDP port by itself. Not only waiting for incoming connection.

**Log Files and Screenshots**
na

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
NXP test drone 
FMUK66 with PX4 master
NavQ companion computer

**Additional context**
My test setup
[image]
QGC is on linux machine in local network as 192.168.0.101
Communication over UDP works only if the UDP communication on Mavlink router towards FMU is setup as normal mode (mav router connects to FMU) and connection from QGC to mavlink-router on TCP port 5790.

Same happens when QGC is connected directly via T1 media converter to FMU. QGC always needs to establish the connection.

dk7xe, 2020-08-20:
@davids5 is this something you can look at please?

dk7xe, 2020-08-20:
hmm.. RTFM.. looks like MAV_BROADCAST set to 1 solves the problem.

dk7xe, 2020-08-21:
Ok, the problem is solved for me if i set the MAV_BROADCAST parameter to 2 - Only multicast

dk7xe, 2020-08-25:
@davids5 - Hi David, now it's an issue again. May you please look at that. Something has happened between between commit df73662 and now.

davids5, 2020-08-25:
@dk7xe - I can get to it the latter half of the week

davids5, 2020-08-27:
@dk7xe See https://github.com/PX4/NuttX/pull/116 once that merges I will do a Firmware update.

dk7xe, 2020-09-08:
Hi @davids5  PX4/NuttX#116 has been merged already. Have u done the FW update?

davids5, 2020-09-08:
@dk7xe yes 

You can see what is on master of Firmware by following https://github.com/PX4/Firmware/tree/master/platforms/nuttx/NuttX

to nuttx, then list commits 

https://github.com/PX4/NuttX/commits/ec20f2e6c5cc35b2b9bbe942dea55eabb81297b6

closed 2020-09-08 by davids5

dk7xe, 2020-09-08:
Thanx David. Meanwhile successfully tested.