PX4/PX4-Autopilot #15582: imu temperature measurement (v1.11.0rc)

NicolasM0, 2020-08-19:
**Describe the bug**
Version hardware: CUAV V5+
Version software: 1.11.0rc and current master (e2e72fd81e2f02c28a786e9fb5099284d0f3bd3e)

IMU temperature measurements are not correct:
- one accel temperature is inconsistent for temperature bellow about 30°C 
- one gyro temperature is nan (for all tested temperature)

It was ok with v1.10

[image]
[image]
[image]

mrpollo, 2020-08-27:
@CUAVcaijie @jinger26 are you aware of any issues?

cuhome, 2020-08-28:
@dagar @davids5 1.11.0 RC firmware does the temperature information of BMI055; the code has this bug.

cuhome, 2020-08-28:
@gitfishup   Please initiate a new merge request to fix it.

gitfishup, 2020-08-28:
You can see this link https://github.com/PX4/Firmware/pull/15638. I have submitted a pull request. @NicolasM0

gitfishup, 2020-08-28:
It works well.
[image]

dagar, 2020-08-28:
> one accel temperature is inconsistent for temperature bellow about 30°C

Looks like the BMI055 accel, I'll check the driver.

> one gyro temperature is nan (for all tested temperature)

That's because the BMI055 (and BMI088) gyro doesn't have a temperature sensor.

dagar, 2020-08-28:
I believe I have a fix for the BMI055 accel temperature. https://github.com/PX4/Firmware/pull/15640

For the BMI055 gyro in the past the temperature it reported was actually being pulled from the accel. In the new driver the accel and gyro are scheduled separately since they're effectively 2 separate sensors (each has a chip select, drdy, slightly different output data rate, etc). 

If the lack of temperature in the BMI055 gyro is a problem for either temperature calibration or an IMU heater I was thinking of adding explicit support for those functions to respect other temperature sources.

dagar, 2020-08-28:
I've fixed the accel temperature reading in https://github.com/PX4/Firmware/pull/15640.

With regards to the BMI055 gyro lack of temperature sensor I'm not sure it's much of a problem. If you look in my last comment (https://github.com/PX4/Firmware/pull/15640#issuecomment-683125028) you'll see that the BMI055 gyro is actually quite stable with respect to temperature.

closed 2020-08-28 by dagar

NicolasM0, 2020-08-31:
> I've fixed the accel temperature reading in #15640.
> 
> With regards to the BMI055 gyro lack of temperature sensor I'm not sure it's much of a problem. If you look in my last comment ([#15640 (comment)](https://github.com/PX4/Firmware/pull/15640#issuecomment-683125028)) you'll see that the BMI055 gyro is actually quite stable with respect to temperature.

Onboard thermal calibration never stop (I think it is linked to this nan temperature)

dagar, 2020-08-31:
> Onboard thermal calibration never stop (I think it is linked to this nan temperature)

If you're in a position to test that quickly let's update it to ignore these cases. In general we really can't assume everything has a valid temperature.