PX4/PX4-Autopilot #15590: Could not find px4 using the following names:

saad2121, 2020-08-20:
when i am running the command 
```
make posix_sitl_default gazebo
```
I am getting error as below.

CMake Error at CMakeLists.txt:31 (find_program):
  Could not find px4 using the following names:

-- Configuring incomplete, errors occurred!
See also "/home/saad/catkin_quad/src/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/saad/catkin_quad/src/Firmware/build/posix_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
[160/592] Building CXX object src/lib/...iles/controllib.dir/BlockLowPass.cpp.o
ninja: build stopped: subcommand failed.
Makefile:148: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

closed 2020-08-20 by TSC21

bamboosdu, 2020-09-27:
Did you solve the problem? I meet the same problem.
If you solved it, could you share me with your solution.

shonigmann, 2020-10-08:
for what it's worth, I ran into this issue after running "make clean"; couldn't find a direct solution but reinstalling PX4 (per https://dev.px4.io/v1.9.0/en/setup/dev_env_linux_ubuntu.html) fixed things for me

Jaeyoung-Lim, 2020-10-08:
The problem is that you have the firmware directory inside catkin_ws

sycamore-yu, 2024-07-28:
I meet the same problem too,the key of the problem is the version of cmake.The profile recommand cmake version from 3.10to 3.16.I change my cmake version from 3.27to 3.15 and problem sovled.The way change cmake version is build soft connection