PX4/PX4-Autopilot #15591: Cube Orange + External Here2 doesn't recognize external compass

DonLakeFlyer, 2020-08-20:
* Current master
* `make cubepilot_cubeorange_default`
* Flashed to board
* QGC Compass Cal
* Only the internal compass shows up

dagar, 2020-08-20:
Is this the GPS1 port with the safety button and lights?

julianoes, 2023-04-03:
@DonLakeFlyer or anyone else: if you have a change, could you re-test with `main`?

I would like to fix or close this but unfortunately I don't have a Here2.

julianoes, 2023-04-03:
Related: https://github.com/PX4/PX4-Autopilot/issues/16070

julianoes, 2023-04-27:
I now have a Here2 and I'm looking into it.

julianoes, 2023-04-27:
Ok, it works for me on `main` (b8b80e409f7a84f885522d54e764b6b4d3d1e8f5):

```
nsh> listener sensor_mag

TOPIC: sensor_mag 2 instances

Instance 0:
 sensor_mag
    timestamp: 57358497 (0.014638 seconds ago)
    timestamp_sample: 57357364 (1133 us before timestamp)
    device_id: 592905 (Type: 0x09, I2C:1 (0x0C))
    x: -0.01200
    y: -0.14700
    z: -0.46650
    temperature: nan
    error_count: 0

Instance 1:
 sensor_mag
    timestamp: 57355139 (0.019321 seconds ago)
    timestamp_sample: 57355107 (32 us before timestamp)
    device_id: 589858 (Type: 0x09, SPI:4 (0x00))
    x: 0.03450
    y: 0.03000
    z: -0.37650
    temperature: nan
    error_count: 0
```

```
dmesg
HW arch: CUBEPILOT_CUBEORANGE
PX4 git-hash: b8b80e409f7a84f885522d54e764b6b4d3d1e8f5
PX4 version: 1.14.0 0 (17694720)
PX4 git-branch: main
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 3f77354c0dc88793a47ff3b57595195ab45f7ba9
...
ak09916 #0 on I2C bus 1 address 0xC
...
```

And v1.13.3:

```
nsh> listener sensor_mag

TOPIC: sensor_mag 2 instances

Instance 0:
 sensor_mag
    timestamp: 183394679 (0.002776 seconds ago)
    timestamp_sample: 183393551 (1128 us before timestamp)
    device_id: 592905 (Type: 0x09, I2C:1 (0x0C))
    x: 0.0315
    y: -0.4515
    z: -0.1230
    temperature: nan
    error_count: 0

Instance 1:
 sensor_mag
    timestamp: 183393380 (0.014204 seconds ago)
    timestamp_sample: 183393348 (32 us before timestamp)
    device_id: 589858 (Type: 0x09, SPI:4 (0x00))
    x: 0.3180
    y: -0.0855
    z: -0.4245
    temperature: nan
    error_count: 0
```

```
dmesg
HW arch: CUBEPILOT_CUBEORANGE
FW git-hash: 1c8ab2a0d7db2d14a6f320ebd8766b5ffaea28fa
FW version: Release 1.13.3 (17630207)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 4a1dd8680cd29f51fb0fe66dcfbf6f69bec747cf
Build datetime: Mar 14 2023 01:22:09
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
...
ak09916 #0 on I2C bus 1 address 0xC
...
```

With that I'm claiming this is working as it should and closing this issue.

closed 2023-04-27 by julianoes